LazeMSS/OctoPrint-UICustomizer #82: ! Error loading webcam

theo78825, 2021-02-14:
This seems to affect the Safari web browser in particular. When loading the Octoprint page, an error saying "! Error loading webcam" displays where the webcam view should be. By going to the "Control" tab, there are no problems viewing the stream there. After a few minutes, the webcam view on the main page eventually loads.
[image]
[image]

LazeMSS, 2021-02-17:
I can't reproduce this - I dont have a stable mac for testing it - does the developer console of Safari browser say anything. It might be handling/reloading of the webcam feeds when the browser/tab it out of "focus".

Can you try and test a bit more/explain - does it only happen in Safari?

theo78825, 2021-02-18:
Will take a look and report back!

theo78825, 2021-02-19:
Here are the errors in the developer console. When I switch to a different tab and come back, it seems to load immediately. 
[image]

LazeMSS, 2021-02-19:
Okay - strange. I will try and fire up my old mac and have a look in the weekend.

LazeMSS, 2021-02-21:
I can't recreate in my safari 13.1.3 (OS: Catalina 10.15.7)

Do you use HLS or normal mjpeg?

theo78825, 2021-02-22:
I’m not sure, sorry. I believe mjpeg.

bergdahl, 2021-02-22:
I see the same problem on my Mac running Chrome. I have 2 OCtopi instances, one running a local webcam and the other running a networks tream. Same problem on both. When I load the page the video is not show, If I switch to another tab and back it displays normally. Chrome reports Cancelled on load though. 

[image]

bergdahl, 2021-02-22:
This is the result of loading the main screen for the instance using an external stream. (Chrome)
[image]

This is the result of loading the main screen for the instance using a local USB webcam. (Safari)
[image]

LazeMSS, 2021-02-23:
Whare are the formats of the streams ?
Can I have a url for the external stream so I can try it on my version.

I have tried in both firefox, chrome, and safari on my own installation - and no problems. I'm using the "standard" webcam option of octopi using mjpeg streamer

bergdahl, 2021-02-23:
Internal stream is the bundled Octopi streamer, so using mjpeg-streamer. External stream is mjpeg,url is not available outside my LAN.

LazeMSS, 2021-02-23:
Okay - I have tried with both - I'm recoding the entire webcam part - I have learned alot since I made that part so maybe the new stuff can fix it - I post a test link for you guys as soon its ready.

LazeMSS, 2021-02-23:
Damm its hard when I can't recreate the error - @bergdahl would it be possible with a remote teamviewer etc. so I could have a look.

"I can also report that the webcam is loading if I check "Hide main camera". - does this mean everything is working fine when this is checked?

bergdahl, 2021-02-23:
Yes, it works as expected if "Hide main camera" is checked.

LazeMSS, 2021-02-23:
@bergdahl  and  @theo78825  please try the develop and confirm if its working or not:

```
Open Settings, then click Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-UICustomizer/archive/develop.zip
Click Install
```

bergdahl, 2021-02-23:
Confirmed.

theo78825, 2021-02-23:
Nice job! Looks good on my end!

commit e63a5e0d (2021-02-23): Hopefully fixed #82
closed 2021-02-24 by LazeMSS