LazeMSS/OctoPrint-UICustomizer #85: GCode Viewer in Main tab zoom error

plorimer, 2021-02-17:
## Description
**
When the GCode Viewer tab is selected, and "zoom in on model" ticked, the right (vertical) scroll bar overlaps the image of the code. This does not happen when UICustomizer is disabled and the GCode viewer uses the standard Octoprint viewport. Wondering if padding your viewport will correct this.

[image]
**
<hr>

**Plugins installed**
- Backup Scheduler[backupscheduler] v0.0.4
- Cancel Objects[cancelobject] v0.4.4
- Cost Estimation[costestimation] v3.2.0
- Dashboard[dashboard] v1.17.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.25.4
- Filament Manager[filamentmanager] v1.7.2
- Fullscreen Plugin[fullscreen] v0.0.6
- ipOnConnect[ipOnConnect] v0.2.4
- MultipleUpload Plugin[MultipleUpload] v1.1.0
- Octolapse[octolapse] v0.4.1
- Preheat Button[preheat] v0.6.0
- PrintJobHistory[PrintJobHistory] v1.10.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Prusa MMU2 Select Filament[mmu2filamentselect] v1.0.3
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.4
- Python 3 Check[python3plugincompatibilitycheck] v0.1.5
- Resource Monitor[resource_monitor] v0.3.1
- SpoolManager Plugin[SpoolManager] v1.0.0
- **UI Customizer[uicustomizer] v0.1.2.8**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.3
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0.3 Safari/605.1.15

LazeMSS, 2021-02-17:
I think this is fixed in the next release

LazeMSS, 2021-02-17:
Should be fixed in the latest release - please try again:
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.0

Reopen if not fixed - thanks.

closed 2021-02-17 by LazeMSS