LazeMSS/OctoPrint-UICustomizer #89: SUper hero Nav bar temp icons

PhilMaddox, 2021-02-20:
## Description
**ENTER DESCRIPTION HERE
With super hero ( the others are fine) the Navbar temp icons are overlaid on the temp data 
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.5.1**
- TouchUI[touchui] v0.3.17
- Navbar Temperature Plugin[navbartemp] v0.14

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.3
- Python 3.7.3
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/88.0.4324.182 Safari/537.36

LazeMSS, 2021-02-20:
@PhilMaddox  I will look into - but you can also try my alternative to toptemp: https://plugins.octoprint.org/plugins/toptemp/
😄

LazeMSS, 2021-02-21:
I think I will remove the super hero theme

LazeMSS, 2021-02-21:
Removed

closed 2021-02-21 by LazeMSS

PhilMaddox, 2021-02-21:
Pity but thanks for taking action

LazeMSS, 2021-02-21:
The new themes are quite similar 🤗