LazeMSS/OctoPrint-UICustomizer #90: Resize of files section

Spooky-Wookie, 2021-02-20:
First things first:
Thx for this great piece of software...
But:
is the any possibility to resize the files section?
Actual it shows only a small overview of the files uploaded to OPi.
[image]
About 50% of the column is empty (at least in my case).
Would love to see an option to fill up the emty space with a bigger files-section view.
[image]

LazeMSS, 2021-02-20:
Great idea :)

LazeMSS, 2021-02-21:
Added in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.2

Closing - please reopen if not working/fixed as expected

closed 2021-02-21 by LazeMSS