LazeMSS/OctoPrint-UICustomizer #91: Many issues with basically all themes.

lightmaster, 2021-02-20:
Found some bugs in the new themes, tested on Firefox. After each theme change, a Ctrl+F5 was used to make sure no CSS was carried over from one theme to the next.

### All Themes:

- Started listing this under each one, until I realized it actually effects every theme except Blue Moon. The Nav Bar shows as the same light gray for almost every theme.

### Blue Moon Theme:

- Are the icons and username in the upper right supposed to be so hard to read?

[image]

### Cyborg Theme:

- Nav Bar is light gray, whereas the small thumbnail in settings shows it being much darker.

[image]

### Slate:

- Nav Bar is partially light Gray, partially dark gray.
- The Dark Gray parts, which I assume is the color the Nav Bar is supposed to be, the icons and text are hard to read
- Hovering over the icons with the dark gray background makes their background turn light gray like the rest of the nav bar

[image]

### Superhero:

- Nav Bar is light gray
- Header and footer text are larger than they should be
- Cancel button is misaligned. I am using the InlineConfirm plugin, but this button looked ok with the other themes.
- DisplayLayerProgress text too high, probably top justified instead of centered?
- Possibly a symptom of the font sizes being too large, but the header and footer bar are about twice as tall as they should be.

[image]

LazeMSS, 2021-02-20:
Thanks I will look ASAP

LazeMSS, 2021-02-21:
It looks fine on my Firefox - I had that issues but it should be fixed in the final release. Im running firefox portable 85.0 (64-bit) - its works with or without caching. From the developer console, Network tab try checking "Disable cache" but it should read the css file as a completly different file

Regarding SuperHero - is how its designed by the author - I know alot of plugins rely on the toolbar being 40px and I will probally fix it.

lightmaster, 2021-02-21:
> Regarding SuperHero - is how its designed by the author - I know alot of plugins rely on the toolbar being 40px and I will probally fix it.

Well that's a bummer. It looks really close to a Themify theme that I liked, and was hoping to be able to drop that plugin entirely if it was also covered by this plugin. The massively blowout fonts on the header and foot and kinda kill it for me. There's already a lot of wasted space in OctoPrint and 90% of my efforts to customize it are just to utilize all the space available.

I'll keep poking with it and see if I can figure out what's going on with the other bits.

LazeMSS, 2021-02-21:
@lightmaster  I will make a orange/black (Prusa) inspired theme - its planned :)

LazeMSS, 2021-02-21:
btw. I'm guessing that the problems you are seeing could be the nginx port forward etc. caching the "active.css" file

lightmaster, 2021-02-21:
Really any good, dark night theme with some nice accenting would be awesome.

I'm 99% sure that I disabled caching of anything for octoprint in the reverse proxy. I will double check though

LazeMSS, 2021-02-21:
Closing now - have tested on 3 computers with firefox.

lightmaster, 2021-02-25:
I've made sure that my reverse proxy has `proxy_cache off;` set, and from my understanding nginx requires you to specifically setup caching, which I haven't.

With the slate theme, I was able to fix the Nav Bar background and text on the right side by playing around with the Inspect windows and disabling all the CSS circled in red below.

Fix Slate Theme Nav Bar Background:
[image]

Fix Slate Theme Nav Text:
[image]

LazeMSS, 2021-02-25:
@lightmaster are you still having problems with this and in which browsers?

lightmaster, 2021-02-25:
Yup, still seeing this on both Firefox and Chrome. The new theme Red Night looks great, and Default (unsurprisingly) looks fine, but all other themes including the Prusa one have jacked up Nav Bars. Its like the Default Nav Bar background is showing up on top of theme, except for Slate where the button on the right side of the Nav Bar have the correct background, but the rest of the Nav Bar looks like the Default background. And since Red Night has its dark background on the Nav Bar, I would think that would rule out any persistent caching issues too.

LazeMSS, 2021-02-25:
Will look now

LazeMSS, 2021-02-25:
What does this return from a developer console?
$('link[href^="/static/webassets/packed_core.css"][rel="stylesheet"]').length
vs
this:
$('link[href^="./static/webassets/packed_core.css"][rel="stylesheet"]').length

lightmaster, 2021-02-25:
both spit back `0`

lightmaster, 2021-02-25:
If I change it to `$('link[href^="/octoprint/static/webassets/packed_core.css"][rel="stylesheet"]').length` I get `1`

LazeMSS, 2021-02-25:
So you are running octoprint in a folder :)

$('link[href*="static/webassets/packed_core.css"][rel="stylesheet"]').length

Does this return 1?

lightmaster, 2021-02-25:
I'm running Octoprint through OctoPi and wherever it defaults to having OctoPrint installed at. My reverse proxy uses `proxy_set_header X-Script-Name /octoprint;` so that OctoPrint can be located at `example.com/octoprint`.

And yes, that last one does return 1

LazeMSS, 2021-02-25:
Okay I will fix it now - develop release inbound.

lightmaster, 2021-02-25:
Curious why one theme worked fine (at least as far as I could tell), but the rest didn't.

LazeMSS, 2021-02-25:
@lightmaster  because the developer of that theme took the time to manullay overwrite all the hardcode octoprint stylesheets.

My plugin normally on a normal setup remove all these hardcode css but your special case don't work so now that is fixed:

https://github.com/LazeMSS/OctoPrint-UICustomizer/archive/develop.zip

Please test.

LazeMSS, 2021-02-25:
Ohh and there is a small bug with the selection of themes in the develop version.

lightmaster, 2021-02-25:
install now

lightmaster, 2021-02-25:
I assume the bug you mention is it not showing you which theme you've selected?

Its amazing how much better these themes look when all the assets load, lol 🤣

With Cyborg and Prusa, I'm seeing that DisplayLayerProgress is much higher than the rest of the text in the Nav Bar, whereas Cosmos and Flatly have DisplayLayerProgress too low.
[image]
[image]

I also notice that Blue Moon and Slate have a nice gap between TopTemp and DisplayLayerProgress which none of the other themes have.

Nice Gap
[image]

No Gap
[image]

LazeMSS, 2021-02-25:
its quite hard to make css that works with everybody elses idea of implementing css - thats why you might see different results.

Again feel free to make you own custom css under advanced or submit pr

LazeMSS, 2021-02-25:
I found the bugs - thanks for reporting and sorry for sounding a bit grumpy - been a long night 😫

LazeMSS, 2021-02-25:
Try again by updating develop

lightmaster, 2021-02-25:
The `line-height` adjustments make the DisplayLayerProgress look a lot better.

Sidenote, thanks for the tip earlier about disabling cache in the Network window, Firefox was very insistent about storing the cache for each and every freaking theme.

LazeMSS, 2021-03-05:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.4
Please reopen/comment it not 😄

closed 2021-03-05 by LazeMSS