LazeMSS/OctoPrint-UICustomizer #95: Top Icon for Firmware Updater not working correctly

Christoph512, 2021-02-21:
If i press the Top Icon for Firmware Updater i get the last opened settings page and not the page for Firmware Updater.

[image]

LazeMSS, 2021-02-21:
And this does not happen when disabling UI customizer?

What have you tried?

Christoph512, 2021-02-21:
When UI Customizer is disabled, it opens the settings page for Firmware Updater, even if i was before on another settings page.

LazeMSS, 2021-02-22:
This is fixed in the next release - please test/try by following the below guide:

Open Settings, then click Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-UICustomizer/archive/develop.zip
Click Install

Christoph512, 2021-02-22:
I tested it along your guide, now its working. Thank you very much! 👍

LazeMSS, 2021-02-22:
Keeping it open until public released :)

LazeMSS, 2021-02-24:
Now released and fixed 👍

closed 2021-02-24 by LazeMSS