LazeMSS/OctoPrint-UICustomizer #96: Top menu text for Navbar Temperature plugin while on dark themes are hard to see

NovaViper, 2021-02-22:
## Description
**ENTER DESCRIPTION HERE
Describe your problem? When using the themes included in the plugin, I notice that the text from the Navbar Temperature plugin do not change and stay black, which makes them really hard to see while on dark themes.

What is the problem? The text for the Navbar Temperature plugin do not change
Can you recreate it? Yes
Did you try disabling plugins? No
Did you remember to update the subject? Yes
<hr>

**Plugins installed**
- Autoscroll[autoscroll] v0.0.3
- Backup Scheduler[backupscheduler] v0.0.4
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.25.4
- DryRun Plugin[DryRun] v1.2.0
- Exclude Region[excluderegion] v0.3.0
- InlineConfirm[inlineconfirm] v1.0.1
- Marlin GCode Documentation[marlingcodedocumentation] v0.12.0
- Navbar Temperature Plugin[navbartemp] v0.14
- OctoKlipper[klipper] v0.3.2
- Printoid Plugin[printoid] v1.0.2
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.4
- SpoolManager Plugin[SpoolManager] v1.3.3
- System Command Editor[systemcommandeditor] v0.3.4
- Tasmota[tasmota] v1.0.1
- TimeToFilament Plugin[TimeToFilament] v1.2.1
- **UI Customizer[uicustomizer] v0.1.5.2**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
Vivaldi v3.6.2165.40
- Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/88.0.4324.186 Safari/537.36

LazeMSS, 2021-02-23:
Yes its hard to support all the different plugins - I will try and add a rule for navbar temp 

Shameless plug: 
Until then you can have a look at my alternative solution top navbar temp: https://plugins.octoprint.org/plugins/toptemp/

NovaViper, 2021-02-23:
I actually went and got the Top Temp plugin, I like it better than the original one I was using. Thanks for making it :D

LazeMSS, 2021-02-24:
Now released and fixed 👍

closed 2021-02-24 by LazeMSS