LazeMSS/OctoPrint-UICustomizer #99: [Bug] Themed compressed file list hides some dropdown settings

lightmaster, 2021-02-25:
When using a theme which compresses the file list so it doesn't show the full file list if it's empty, this causes the bottom of the drop down box to be hidden when the file list is the last item in a column. In my screenshot, the `Hide successfully printed files` option is inaccessible. I can go into the Inspect window and delete the HTML for the footer bar, and then I can see and interact with this last option again.

[image]

LazeMSS, 2021-02-25:
What themes makes the file folder that small?

LazeMSS, 2021-02-25:
Fixed in develop

lightmaster, 2021-02-25:
Installing it again to test now.

lightmaster, 2021-02-25:
Well, it does show up now, but I thought the compressed empty file list was a feature, not a bug. IMO, it'd be nicer to have the list compressed when empty/low, and when menu is opened, there's invisible padding added to the outside of the dropdown menu such that it forces the page to scroll down lower.

lightmaster, 2021-02-25:
my temp fix was custom CSS

```CSS
.accordion-heading .dropdown-menu {
  padding: 5px 0 40px 0;
}
```
However that creates a visible padding on the bottom of the dropdown menu. idk much about CSS and couldn't figure out what to do to make invisible padding.

LazeMSS, 2021-02-25:
@lightmaster  feel free to make these changes and do a pull request.

LazeMSS, 2021-02-25:
Ohh and I made it to big - fixed now

lightmaster, 2021-02-25:
Thankfully the `min-height: 70px` looks good to me, cuz I've hit the extent of my knowledge of CSS and unable to create a transparent padding that doesn't jack up the menu.

With the new height, I am unable to get the menu to sink below the foot now.

LazeMSS, 2021-03-05:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.4
Please reopen/comment it not 😄

closed 2021-03-05 by LazeMSS