LazeMSS/OctoPrint-UICustomizer #101: layout items not hiding

redkurn, 2021-02-26:
## Description
**ENTER DESCRIPTION HERE layout items not hiding
Describe your problem? layout items not hiding
What is the problem? printer notifications not hidden
Can you recreate it? yes, click hide and watch it pop up.
Did you try disabling plugins? none needed to disable
Did you remember to update the subject?**
<hr>

**Plugins installed**
- Autoscroll[autoscroll] v0.0.3
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- BetterHeaterTimeout[BetterHeaterTimeout] v1.3.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.25.4
- GcodeEditor[GcodeEditor] v0.2.12
- Octoprint-Display-ETA[display_eta] v1.1.3
- PrettyGCode[prettygcode] v1.2.4
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Terminal Messaging[terminalmessaging] v0.1.0
- Themeify[themeify] v1.2.2
- Top Temp[toptemp] v0.0.1.1
- **UI Customizer[uicustomizer] v0.1.5.3**

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: false
- compressTempControls: false
- disableTermInactive: true
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:86.0) Gecko/20100101 Firefox/86.0

LazeMSS, 2021-02-26:
Well if there is a notification then it will show up again.

redkurn, 2021-02-26:
Until I completely disabled notification popups I was having a glorious time...
I clicked to hide it, am I not understanding the setting properly?

[image]

LazeMSS, 2021-02-26:
The notification you can hide with my plugin are not these
The notifications my plugin can hide is part of the main ui.

The notifications you are seeing are probably related to another plug-in and it also appears like your printer is sending rubbish output

closed 2021-02-27 by LazeMSS