LazeMSS/OctoPrint-UICustomizer #104: Top Icons in compact mode and Cyborg Theme

arut16, 2021-02-27:
## Description
Top Icons in compact mode are over temperatures with Cyborg Theme. And it will be nice if icons and text can be changed in white for more visibility.

[image]

**Plugins installed**
- Z Probe Offset[z_probe_offset] v0.5
- **UI Customizer[uicustomizer] v0.1.5.3**
- Access Anywhere - The Spaghetti Detective[thespaghettidetective] v1.6.0
- Status Line[status_line] v0.1.1
- OctoPrint-WiFiStatus[wifistatus] v1.5.1
- Cura Thumbnails[UltimakerFormatPackage] v0.2.2
- TimeToFilament Plugin[TimeToFilament] v1.2.1
- Tab Order[taborder] v0.5.12
- Resource Monitor[resource_monitor] v0.3.1
- PSU Control[psucontrol] v0.1.11
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Printoid Plugin[printoid] v1.0.2
- Navbar Temperature Plugin[navbartemp] v0.14
- M117NavBar[M117NavBar] v0.1.1
- GcodeEditor[GcodeEditor] v0.2.12
- Exclude Region[excluderegion] v0.3.0
- BLTouch Plugin[BLTouch] v0.3.4
- Octolapse[octolapse] v0.4.1
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.25.4
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- Backup Scheduler[backupscheduler] v0.0.4
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: false
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: false
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- Octoprint 1.5.3
- Python 3.7.3
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/88.0.4324.190 Safari/537.36

JimKnightster, 2021-02-28:
Agreed, I have tried most every theme available and they are all very hard to read.

LazeMSS, 2021-02-28:
I have have some fixes to it - but please also try my toptemp plugin:

https://plugins.octoprint.org/plugins/toptemp/

LazeMSS, 2021-02-28:
Ahh I noticed the fix I made earlier for this doe not work 👎

arut16, 2021-02-28:
> I have have some fixes to it - but please also try my toptemp plugin:
> 
> https://plugins.octoprint.org/plugins/toptemp/

Perfect! Thanks ! All problems solved for me !

JimKnightster, 2021-03-02:
> I had some fixes to it - but please also try my toptemp plugin (shameless plug):
> 
> https://plugins.octoprint.org/plugins/toptemp/

Thanks for the Plugin plug, I like yours better anyway.

LazeMSS, 2021-03-05:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.4
Please reopen/comment it not 😄

closed 2021-03-05 by LazeMSS