LazeMSS/OctoPrint-UICustomizer #107: Latest 1.5.4 doesn't work right

mbourque, 2021-03-05:
After I installed, my three column layout was now just one. Had to roll back the changes. I am using Chrome on OSX.

LazeMSS, 2021-03-05:
So it just broke after updating?

LazeMSS, 2021-03-05:
Can you maybe look in your developer console and look for errors there?

mbourque, 2021-03-05:
I will try again, I reverted back but should had captured more info for you.

LazeMSS, 2021-03-05:
did you revert back to the old version?

mbourque, 2021-03-05:
Yes

mbourque, 2021-03-05:
After my print job finishes I will do it all again and send you screenshots and any errors.

LazeMSS, 2021-03-05:
okay if you try and upgrade let me know and try have a look in the developer console - nothing regarding layout should have happend but there might be a javascript errror causing it all to fail - do you run themify or use any of the colors of octorprint apperance.

reclaimer5146, 2021-03-05:
Can confirm, themeify breaks it. I disabled themeify and it fixed it. Then I enabled again and it broke it again.

What do you need from me? I don't need to downgrade as I have nothing printing right now.

LazeMSS, 2021-03-05:
I will try installing themify

LazeMSS, 2021-03-05:
@reclaimer5146  do you have any custom css in themify?

Hobby4life, 2021-03-05:
> Can confirm, themeify breaks it. I disabled themeify and it fixed it. Then I enabled again and it broke it again.
> 
> What do you need from me? I don't need to downgrade as I have nothing printing right now.

I gues the otherway arround.... UICustomizer broke Themify.
Both worked in harmony before 1.5.4 ;-)

LazeMSS, 2021-03-05:
:)

Hobby4life, 2021-03-05:
LazeMSS wil fix this.. I know he will :-)

LazeMSS, 2021-03-05:
Yes bug found 🥇

LazeMSS, 2021-03-05:
Fixed and released.

closed 2021-03-05 by LazeMSS

LazeMSS, 2021-03-05:
@Hobby4life , @mbourque  and @reclaimer5146  why would anyone run UI Customizer together with Themify 😛

reclaimer5146, 2021-03-05:
@LazeMSS Honestly, I only installed UI Customizer and it laid out perfectly so I didn't mess with Themeify. But now that I'm messing with it, I'll remove Themeify 😛

mbourque, 2021-03-05:
> @Hobby4life , @mbourque and @reclaimer5146 why would anyone run UI Customizer together with Themify 😛

Good question. Maybe I don't need both now that you added themes.

Hobby4life, 2021-03-06:
Thanks for the fix  :-) 👍