LazeMSS/OctoPrint-UICustomizer #111: Top bar background not showing.

Bob90, 2021-03-06:
Hi,

Firstly, what a great UI and Top Temp having customizable tabs is so helpful.

My layout has a missing background image as attached. And I haven't been able to get it appear.
[image]

Do you know what I can do?

I have ArcWelder, Bed Vizualiser, Custom Control Editor, Firmware Check, M73 Progress, OctoKlipper, PrintTimeGenius Plugin, Printer Diagnostics, Printer Notifications, Top Temp and Virtual Printer installed.

At installation I had: TouchUI, Tab Order, Consolidate Temp Control installed. I've disabled them since installing your plugin.

LazeMSS, 2021-03-06:
Have you looked in settings > appearance? Should look like something I have attached
And what theme are you using?

[image]

LazeMSS, 2021-03-08:
@Bob90 ???

Bob90, 2021-03-08:
Hi LazeMSS,

Sorry, this was it. Thank you.

I don't remeeber me doing it. Could another plugin change this not revert on install/unload?

closed 2021-03-08 by Bob90