LazeMSS/OctoPrint-UICustomizer #113: Themes not loading

JaredWith1r, 2021-03-11:
## Description
Trying to change the theme, but it just flashes 'Loading themes...' endlessly. Just me? Any insight?

Thanks.
<hr>

**Plugins installed**
- Consolidated Tabs[consolidatedtabs] v1.0.0
- Enclosure Plugin[enclosure] v4.13.1
- PrintJobHistory[PrintJobHistory] v1.10.0
- Simple Emergency Stop[simpleemergencystop] v1.0.4
- **UI Customizer[uicustomizer] v0.1.5.5**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: false
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: true

**Software versions**
- OctoPrint 1.5.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; WOW64; rv:68.0) Gecko/20100101 Firefox/68.0

LazeMSS, 2021-03-11:
Can you look in your developer console and look for error? 

https://developer.mozilla.org/en-US/docs/Tools/Browser_Console

JaredWith1r, 2021-03-11:
That page loads, though when I hit a Preview, I get the following
"[Exception... "Favicon at "https://lazemss.github.io/favicon.ico" failed to load: Not Found."  nsresult: "0x80004005 (NS_ERROR_FAILURE)"  location: "JS frame :: resource:///modules/FaviconLoader.jsm :: onStopRequest :: line 199"  data: no]"

Additionally, when I go to load themes, I get the following:
"TypeError: addThis.replaceAll is not a function"

JaredWith1r, 2021-03-11:
I am using a work PC for this. I tried a personal computer (with less control, presumably) and I do not have this issue.

SO, feel free to close this issue. Thanks for the support!

LazeMSS, 2021-03-12:
:)

closed 2021-03-12 by LazeMSS