LazeMSS/OctoPrint-UICustomizer #114: Navbar shows a checkerboard / grey pattern when using themes (armbian - OPi)

v-abt, 2021-03-15:
Hi,
I don't know if this is the right place to post this issue or if this may belong into to the themes repo (just let me know, if you want it to be moved).
I'm using your plugin on an armbian machine (OrangePi +2E) and have some weird problems when using themes:
on all themes, escept "blue moon", the navbar gets a grey checkerboard pattern as if something is missing.
[image]

additionally the prusa tribute theme messes up the button layout in the connection panel:
[image]

are there any additional packages necessary to get this plugin working on armbian?
The issue seems to be browser independent (navbar looks the same in the latest firefox, chrome and edge browsers and even on mobile chrome)
[octoprint.log](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/6140174/octoprint.log)

LazeMSS, 2021-03-16:
Have you enabled the transparent options under settings > OctoPrint   > Appearance?

If so disable it

v-abt, 2021-03-16:
Yes I had, disabling it fixed the navbar. Thanks a lot! 

The "Prusa Tribute" theme still has that weirdly placed abort button. 
As it is the only broken theme, I guess there is something wrong with it / conflicting with the preheat and/or inline confirm button.

I'm going to close this ticket now.

closed 2021-03-16 by v-abt