LazeMSS/OctoPrint-UICustomizer #115: Applied Blue moon theme (successfully) but can no longer change the text color up top (like my username).

ChrisWoodruff, 2021-03-18:
## Description
I recently loaded UI Customizer, I like it a lot. I was messing around with the themes and picked Blue moon. I liked it except that navbar temp plugin had a white background. So I went back to default.
All the text on that line (in electric blue) is stuck in electric blue. On a white background it is hard to see (it's washed out). 
I tried uninstalling (including Clean Data), I tried installing Themify but it didn't change it. I uninstalled Themify and reinstalled UI Customizer and it didn't change, even if picking other themes.
I don't mind my name so much, but the temp is rough since I use that regularly.  Thanks!

[image]

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- Cancel Objects[cancelobject] v0.4.4
- CANVAS[canvas] v2.3.3
- GcodeEditor[GcodeEditor] v0.2.10
- M117NavBar[M117NavBar] v0.1.1
- Navbar Temperature Plugin[navbartemp] v0.14
- OctoFlat Theme[octoflat] v0.2.0
- Palette 2[palette2] v2.4.13
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Simple Emergency Stop[simpleemergencystop] v1.0.4
- **UI Customizer[uicustomizer] v0.1.5.5**
- Z Probe Offset[z_probe_offset] v0.5

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: true

**Software versions**
- OctoPrint 1.5.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36

LazeMSS, 2021-03-19:
Have you checked color and Transparent color in Settings > OctoPrint > Appearance?

ChrisWoodruff, 2021-03-19:
No I had not, thanks for letting me know!
Using that I was able to get the font color back to black.
All is now right in the world, er, at least this part of it. Have a
great day!

closed 2021-03-25 by LazeMSS