LazeMSS/OctoPrint-UICustomizer #116: I cant print anything.

thatwolfrio, 2021-03-19:
## Description
**ENTER DESCRIPTION HERE
Describe your problem? So whenever i click the load to print button, it just doesn't do anything. it does the animation for clicking, but it doesn't print.
What is the problem? Whenever i click the load to print button it does nothing.
Can you recreate it? Yes, every time i try to click the button, it doesn't work.
Did you try disabling plugins? Yes
Did you remember to update the subject? I just updated.
<hr>

**Plugins installed**
- Access Anywhere - The Spaghetti Detective[thespaghettidetective] v1.6.1
- AstroPrint[astroprint] v1.6.3
- Bed Leveling Wizard[bedlevelingwizard] v0.2.4
- DisplayProgress[displayprogress] v0.1.3
- Mattacloud Beta[mattacloud] v0.1.2
- OctoEverywhere![octoeverywhere] v0.7.4
- Octolapse[octolapse] v0.4.1
- RaiseCloud[raisecloud] v1.1.0
- Simple Emergency Stop[simpleemergencystop] v1.0.4
- Slicer[slicer] v1.4.4
- Slicer Print Time Estimator[SlicerEstimator] v1.0.0
- Top Temp[toptemp] v0.0.1.2
- **UI Customizer[uicustomizer] v0.1.5.5**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: false

**Software versions**
- OctoPrint 1.5.3
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36

LazeMSS, 2021-03-19:
Can you print without UI Customizer enabled?

LazeMSS, 2021-03-25:
Closing due to no response and lack of info.

closed 2021-03-25 by LazeMSS