LazeMSS/OctoPrint-UICustomizer #117: Latest version messed up the text color of the Navbar in the Default theme

KL-XL, 2021-03-19:
Hi there - I have been using UI Customiser and Themeify concurrently without problems until version UI Customiser 0.1.5.5.  Here is how the Navbar looks like now (UI Customiser theme = Default, Themeify theme = Discoranged):

[image]
This is how it looked like before:
[image]

Note: The PSU Control icon is green because during the first snapshot it was pressed (so it is not an issue). I had to disable UI Customizer to show the previous Navbar formatting so the order of icons has changed, but it is not a concern.

Thank you

LazeMSS, 2021-03-22:
I will look .

LazeMSS, 2021-03-22:
I can't recreate - below is how it looks on my pc - sure you did not enter some custom CSS or maybe cache in the browser.

My plugin disables any themify support if you enable a custom theme - in the next version if will offer to disable themify and other stuff that might conflict

[image]

KL-XL, 2021-03-22:
UI Customizer custom themes are disabled. I reloaded cleaning the cache - no change. No custom css. Here are my Themify settings:
[image]
Disabling UI Customizer fixes the issue.

Here is the list of all plugins installed in case it helps:
|  Access Anywhere - The Spaghetti Detective (1.6.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_thespaghettidetective
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (1.0.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer
|  BLTouch Plugin (0.3.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_BLTouch
|  Consolidate Temp Control (0.1.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_consolidate_temp_control
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Cura Thumbnails (0.2.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_ultimakerformatpackage
|  Dashboard (1.18.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.25.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Filament Manager (1.7.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentmanager
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.2.4) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  Floating Navbar (0.3.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_floatingnavbar
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  HeaterTimeout (0.0.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_HeaterTimeout
|  InlineConfirm (1.1.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_inlineconfirm
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  Navbar Temperature Plugin (0.14) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_navbartemp
|  Octolapse (0.4.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse
|  OctoRelay (1.0.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octorelay
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  Preheat Button (0.7.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_preheat
|  PrintJobHistory (1.10.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory
|  PrintTimeGenius Plugin (2.2.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  PSU Control (0.1.11) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_psucontrol
|  Resource Monitor (0.3.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_resource_monitor
|  Simple Emergency Stop (1.0.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_simpleemergencystop
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  Terminal Commands Extended (0.1.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_terminalcommandsextended
|  Themeify (1.2.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_themeify
|  UI Customizer (0.1.5.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_uicustomizer
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer

LazeMSS, 2021-03-22:
And no custom CSS inside UI Customizer under advanced?
Can you try and look in developer tools and look at the css for the text items that have the text-shadow added.

Btw. is the problem there with text-shadow if you only use UI Customizer?

KL-XL, 2021-03-22:
1. No custom CSS in UI Customizer under advanced
2. Here you go:
[image]
[image]
3. No, if I disable Themeify there is no problem, but obviously the orange theme is gone then. Interestingly, if I apply any of your custom themes, there is no such problem, just with the default one.

LazeMSS, 2021-03-24:
1. Ok
2. There must be a text-shadow somewhere - try browsing up/down the dom
3. Okay

KL-XL, 2021-03-24:
> 2\. There must be a text-shadow somewhere - try browsing up/down the dom

It is on screenshot 2, line 12-13, which is the overall css file that opens when you click on <style> in the 3rd from the bottom right-hand-side box from screenshot 1 (with the 'Settings' button selected).
It looks like the default css is overwritten by UI Customizer both setting the text shadow and text color

LazeMSS, 2021-03-25:
Ahh now i know - its because the default theme is added at the end - I need to rewrite so the default theme is added to the top before other css files.

I might be hard to fix - maybe I need to make a Discoranged theme for UI Customizer :)

KL-XL, 2021-03-25:
Thank you. I will defer to you how to fix it :) As a coder myself I guess the cleanest is to change the order, but I kind of like the idea of adding a dicoranged theme as it will allow me to have just one plugin of the two installed :)

LazeMSS, 2021-03-25:
Take a look at: https://github.com/LazeMSS/OctoPrint-UICustomizerThemes

KL-XL, 2021-03-25:
Yes, I tried them all. I kind of prefer Themeify's Discoranged theme though (dark grey interface with orange highlights) :) In a combination with your Slate I like it even better. 
[image]

LazeMSS, 2021-03-25:
There is no easy fix - I will try and look into a theme. Until then you can use the custom css of UI Customizer and disabled the text-shadow 😸

KL-XL, 2021-03-26:
Thank you. The text-shadow issue only occurs with Default theme. So I think I can get by with Themeify's Discoranged + UI Customizer's Slate at the moment :)
I guess I might write a custom css to modify Slate to look more like Discoranged, but that will have to wait till a get a spare moment :)

LazeMSS, 2021-03-26:
Okay great thanks. More themes are coming.

KL-XL, 2021-03-26:
Great. I look forward to it. Just to make sure everything is captured before we abandon this conversation:
- the problem only seems to occur in the Default theme
- the symptoms I identified and that affect the Navbar are: 
1. text shadow added to text and icons
2. text color is changed to the color of the background (in case of Discoranged it is dark grey from orange)

LazeMSS, 2021-03-26:
I have it on the backlog/todo - not forgotten 😄

closed 2021-03-30 by LazeMSS