LazeMSS/OctoPrint-UICustomizer #119: Sidebar webcam and popout rotation not working

alefchak, 2021-03-24:
The webcam sidebar image is no longer respecting the "Flip webcam horizontally" and "Flip webcam vertically" options in the main Octoprint settings.

I have both of these settings checked, and while the main webcam display appears correct, the sidebar camera is not.

Additionally, if I click on the pop out icon in the corner of either, the large display is also incorrect (upside down).

I have the same issue with all browsers I've tried - Chrome on macOS, Windows, and Android, as well as Safari on macOS.

[image]

LazeMSS, 2021-03-24:
Sorry - I can't replicate this in anyway on my platform.

alefchak, 2021-03-25:
I disabled plugins until I the behavior changed - it appears to be an interaction with the [Dashboard](https://plugins.octoprint.org/plugins/dashboard/) plugin. I don't think it modifies the webcam but it does have an option to add it to its display.

LazeMSS, 2021-03-25:
Maybe they implement something extra on the webcam container etc. that causes my plugin to loose the options. I will try when I have time and either fix it or report it to the dashboard team.

LazeMSS, 2021-03-25:
I tried install the dashboard - i have no problems. What kind of webcam stream are you using?

And can you try and run 
`$("#IUCWebcamContainerInner img").attr('class')`

from the web developer console?

LazeMSS, 2021-03-25:
From my printer 👍 
[image]

My settings:
[image]

alefchak, 2021-03-26:
> I tried install the dashboard - i have no problems. What kind of webcam stream are you using?
> 
> And can you try and run
> `$("#IUCWebcamContainerInner img").attr('class')`
> 
> from the web developer console?

I'm using a Raspberry Pi camera module. I am have `camera="raspi"` and `camera_raspi_options="-x 1296 -y 972 -fps 30 -quality 100"` set in the Octoprint configuration, and the following settings in the web UI:

[image]

`$("#IUCWebcamContainerInner img").attr('class')` is returning `undefined` while the Dashboard plugin is active, but `"flipH flipV"` while it's disabled.

I did notice that if I uncheck the flip image options, save, recheck them, and save, the appropriate `"flipH flipV"` classes are applied, but go away when the page is refreshed.

LazeMSS, 2021-03-26:
Thanks I have no Idea why this is happening but will try and look.

Btw there is a way of flipping the video feed directly in the RPI driver. It will fix your problems for now I think. If you know your way around bash/shell then it might be easier.

LazeMSS, 2021-03-26:
I tried same settings - I never have this problem.

Btw. try this from bash/shell on rpi:
```
sudo service webcamd stop
cd /home/pi/mjpg-streamer
./mjpg_streamer -i "./input_raspicam.so -hf -vf" -o "./output_http.so -w ./www"
```
This should flip the camera etc: https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_raspicam/README.md

Dag0d, 2021-03-26:
I do have a similar problem, the sidebar image is fine but if I enlarge the picture it's upside down. if I check the code, the "img class="flipH flipV"" is missing.

LazeMSS, 2021-03-27:
@Dag0d  what type of cam etc?

Dag0d, 2021-03-27:
@LazeMSS a normal raspberry pi v2 camera

LazeMSS, 2021-03-28:
There should be no difference between the hardware camera types.

@Dag0d  any other plugins installed that might affect the camera?

Dag0d, 2021-03-28:
You are right, I did some testing and the error only occurs when the dashboard plugin is enables as well.

LazeMSS, 2021-03-28:
I can't recreate it no matter what i do - it must be something about the order of plugins loading etc.

@Dag0d + @alefchak can you try and post your plugins so we can look for matches and maybe its a combination of plugins

tutanra, 2021-03-28:
Show in dashboard github, it solves, only occurs when webcam widget in dashboard is disabled.

Dag0d, 2021-03-28:
@tutanra this did the trick!

LazeMSS, 2021-03-28:
> Show in dashboard github, it solves, only occurs when webcam widget in dashboard is disabled.
> 
> [j7126/OctoPrint-Dashboard#268](https://github.com/j7126/OctoPrint-Dashboard/issues/268)
> 
> This is the temporary fix : https://github.com/j7126/OctoPrint-Dashboard/archive/refs/heads/development.zip

Ohh awesome. I made a generic feature request because I noticed your naming did conflict on some items but I was still trying to find the cause. Closing this one when you release it ok?

LazeMSS, 2021-04-08:
@tutanra when will this be released/include in the main ?

LazeMSS, 2021-04-25:
Done - I can't fix this

closed 2021-04-25 by LazeMSS