LazeMSS/OctoPrint-UICustomizer #120: Layout width exceeds 12

seppelicous, 2021-03-25:
Hi,
the layout must not exceed 12 but it does anyway. I set the width to Left 3, Center 5 and Right 4. It works for a few hours, but then, I do not know why, the width of the center changes automatically to "6" and exceeds the value 12. 
This results in a vertically aligned layout.

kind regards,
Sebastian

[image]

LazeMSS, 2021-03-25:
Thats really strange. I will test asap

seppelicous, 2021-03-25:
I just uninstalled the Job History Plugin and after reboot the layout is still as expected. Sounds even more strange

LazeMSS, 2021-03-25:
The settings in the "Layout" settings should be static so it should never be able to be 13 :(
Do you mean the Job History Plugin was making the broken layout?

seppelicous, 2021-03-25:
[image]
Okay after restart, the layout is broken again. Layout ist again 3+6+4 - that is weird.
Should I reinstall the plugin?

LazeMSS, 2021-03-25:
yes or just clear plugin data - thats easier

LazeMSS, 2021-03-25:
I will write fail check to prevent this from happening.

LazeMSS, 2021-04-14:
Closing - should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.6

closed 2021-04-14 by LazeMSS