LazeMSS/OctoPrint-UICustomizer #121: Custom CSS

Revel8804, 2021-03-25:
for the Custom CSS... how do i use it....

i have tried the following. I am not sure what I need to put in there.

`UICfixedHeader UICfixedFooter UICResponsiveMode UICPreviewON {
background-color: #ff69b4
}`

`background-color: #ff69b4`

`#navbar .navbar-inner {
    background-color: #ff69b4;
}`

`.navbar-inner {
    background-color: #ff69b4;
}`

LazeMSS, 2021-03-25:
Not sure what you want - do you want a pink background color on the top and bottom bars?

```
#navbar .navbar-inner,.footer .navbar-inner{
    background-color: #ff69b4 !important;
    background-image:none;
}

```

Then this should work...

LazeMSS, 2021-03-25:
It not working then reopen - thanks.

closed 2021-03-25 by LazeMSS