LazeMSS/OctoPrint-UICustomizer #124: Compatibility with OctoDash and TouchUI

wbatzle, 2021-04-07:
Is there a way to make this compatible with the two plugins Octodash and TouchUI for the 3.5 inch screen mod? I love this interface on all my devices but it keeps crashing and reloading on my dev pi trying to get it to work. I was planning on doing the touch screen mod on my Prusa this weekend but I think I am going to put that on hold until this works with it. Any help you can provide is appreciated.

LazeMSS, 2021-04-07:
Crashing? What do you mean?

I have tested with both plugins previously and there is even separate support for touch UI plugin. But the touch UI and UI customizer can conflict, take a look here: https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/60

So you can force the touch UI to load when you need it..

wbatzle, 2021-04-07:
I can show you if you have time.

LazeMSS, 2021-04-08:
Sure are you on discord?

wbatzle, 2021-04-08:
Yes. I wiped it out. but I created an image of it first. I wanted to be sure it wasn't something I was doing wrong. I am now working on a new image and it's not even getting to that point. I have to figure out what I did to get TouchUI to even load to start with. I am doing this all on my dev pi. So we can wipe it out as much as we need to.

LazeMSS, 2021-04-08:
contact me on discord: https://discord.octoprint.org/

LazeMSS, 2021-04-09:
@wbatzle  did you fix it?

LazeMSS, 2021-04-12:
Ping @wbatzle ?

wbatzle, 2021-04-12:
I am closing this issue. I was able to get it working. I am not sure what happened but 5th time is a charm. I will delete the image with it being broken at some point. I am not sure what happened. Did everything the same each time.

closed 2021-04-12 by wbatzle