LazeMSS/OctoPrint-UICustomizer #125: Tabs can't be reordered on Firefox

colejd, 2021-04-08:
## Description
**Tabs can't be reordered on Firefox. Clicking and dragging does nothing. I've verified that it works on Chrome on the same machine.**
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.5.5**
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Navbar Temperature Plugin[navbartemp] v0.14
- OctoPrint-IFTTT[IFTTT] v1.3.1
- HeaterTimeout[HeaterTimeout] v0.0.3
- GcodeEditor[GcodeEditor] v0.2.12
- FileManager[filemanager] v0.1.6
- Dashboard[dashboard] v1.18.1
- Macro[macro] v0.3.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.26.0
- Bed Visualizer[bedlevelvisualizer] v1.0.1
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: false

**Software versions**
- OctoPrint 1.5.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10.16; rv:86.0) Gecko/20100101 Firefox/86.0

LazeMSS, 2021-04-09:
Yes there seems to be a problem - it kinda works if you click and hold before start draging - but it needs to be fixed i agree.

LazeMSS, 2021-04-14:
Closing - should be fixed by: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.6

closed 2021-04-14 by LazeMSS