LazeMSS/OctoPrint-UICustomizer #126: All Grey Theme

mac-za, 2021-04-09:
Morning

I could not find the correct forum for a suggestion:

Could you assist with an all grey theme? The cyborg theme partially works, but I would like it if the temperature graph background could also be grey. Unless I am missing something for tweaking the colour myself?

Thanks in advance

Kenneth

LazeMSS, 2021-04-09:
There are more themes coming - promise :)

LazeMSS, 2021-04-09:
ohh the temperature graph in the cyborg theme - that should already be black/dark - or is it in the toptemp plugin?

mac-za, 2021-04-09:
Hi, yes

If the temp graph can also be grey please.

I like the "dark mode" and the temp graph is very bright otherwise.

Thanks for the great work

Kenneth

LazeMSS, 2021-04-09:
Can you post a screenshot?

mac-za, 2021-04-09:
Hi
Will do so a bit later tonight. Thanks

mac-za, 2021-04-12:
Hi

I re-activated the plugin to show you the white background I was
experiencing on the Temperature Graph.

I already had the Themeify plugin installed.

The graph now has a grey background.

I have select the Cyborg and Slate themes with the same results,

Kind Regards

Kenneth

LazeMSS, 2021-04-12:
@mac-za themeify does not apply themes to the popovers and also might have problems together with my plugin.

Could you try using one the UI Customizer themes with Themify disabled?

mac-za, 2021-04-12:
Hi

I'll look at that tomorrow morning

Thanks for your replies.

mac-za, 2021-04-13:
Morning

I have disabled the Themeify app.

I selected the cyborg theme and have attached a screenshot

[image: image.png]

Would it be possible to have a grey background?

I have re-enabled Themeify and this is the result

[image: image.png]

Thanks again for your responses

Kenneth

LazeMSS, 2021-04-13:
You cant include images in email replies :(

mac-za, 2021-04-13:
Hi

I've added two images two my google drive folder

The link is below

https://drive.google.com/drive/folders/1nrvpXBdz8CwsgfVWE4hvabGobbrMSr2y?usp=sharing

LazeMSS, 2021-04-13:
You are using some kind of plugin for the graphs - that not the standard temperature graphs.

LazeMSS, 2021-04-14:
@mac-za  my guess is this plugin:
https://plugins.octoprint.org/plugins/tempsgraph/

My themes don't currently support this - sorry

mac-za, 2021-04-14:
Hi,

I did not reply.

I am using the temps graph plugin.

Thanks again for your help

Kenneth

mac-za, 2021-04-14:
Hi again

Meant to say, sorry for not replying earlier.

Thanks again

LazeMSS, 2021-04-14:
@mac-za  you can just set the colors using the custom settings for colors

LazeMSS, 2021-04-17:
Closing, but on the backlog - this is not 100% related - its hard to support all plugins colors etc in my themes.

closed 2021-04-17 by LazeMSS