LazeMSS/OctoPrint-UICustomizer #128: SoC temperature reading is cut off

npburns224, 2021-04-11:
## Description
**Temperature reading for the SoC in the navbar is cut off when enabling the option "Compact "Navbar Temperature Plugin". I can only see the first digit past the decimal point.**
<hr>

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.0.1
- Cancel Objects[cancelobject] v0.4.4
- Change Filament Plugin[Change_Filament] v0.3.2
- FileManager[filemanager] v0.1.6
- Floating Navbar[floatingnavbar] v0.3.6
- Navbar Temperature Plugin[navbartemp] v0.14
- OctoPod Plugin[octopod] v0.3.0
- Preheat Button[preheat] v0.7.2
- PrettyGCode[prettygcode] v1.2.4
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Simple Emergency Stop[simpleemergencystop] v1.0.4
- **UI Customizer[uicustomizer] v0.1.5.5**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: true

**Software versions**
- OctoPrint 1.5.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:87.0) Gecko/20100101 Firefox/87.0

LazeMSS, 2021-04-12:
Try this: https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/59#issuecomment-754834062

LazeMSS, 2021-04-12:
Closing - this can be fixed a described https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/59#issuecomment-754834062

closed 2021-04-12 by LazeMSS