LazeMSS/OctoPrint-UICustomizer #129: Plugin not behaving as intended

ZeekWorld, 2021-04-13:
Not sure if other plugins are taking over but since the last batch of plugins updates the ui got all messed up.
Tabs on top are set to only show icons, but they are showing text, there is a new widget in the center of the screen "General" Spool, Spool Vendor, etc I can't get rid of. Right side widget have shifted to the left column....

Here are the pictures..

[image]

[image]

[image]

ZeekWorld, 2021-04-13:
OctoPrint 1.6.0rc1 Python 3.7.3 OctoPi 0.18.0

LazeMSS, 2021-04-13:
Could this be related to https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/134

ZeekWorld, 2021-04-13:
Thank you, I will disable the Print History plugin and see how it goes.

yourbuddydinec, 2021-04-13:
Same thing was happening to me when I updated the plugins it said needed it. Turning off PrintJobHistory fixed it so seems to be a problem between the two.

ZeekWorld, 2021-04-13:
Yep, disabling PrintJobHistory fixes the UI

LazeMSS, 2021-04-14:
Closing should be fixed now: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.6

closed 2021-04-14 by LazeMSS