LazeMSS/OctoPrint-UICustomizer #130: OctoPrint-UICustomizer plugin issues after installing update PrintJobHistory 1.12.dev1

David-Maisonave, 2021-04-14:
Ran into OctoPrint-UICustomizer (**0.1.5.5**) plugin issues after installing update **PrintJobHistory 1.12.dev1**.
Before the update, the UI plugin was working with no problems.  Immediately after PrintJobHistory update, the control buttons on [Temperature]  and [Control] tabs were disabled, and the UI plugin could not control the tab order.

Enable PrintJobHistory and disable OctoPrint-UICustomizer, then the control button **starts** working.

Enable PrintJobHistory and enable OctoPrint-UICustomizer, then the control button **stops** working.

Disable PrintJobHistory and enable OctoPrint-UICustomizer, then the control button **starts** working.

With [PrintJobHistory1.12.dev1] disabled, the OctoPrint-UICustomizer plugin starts working again.

They were working together with version Print-Job-History-1.10.0, but I'm not sure how to revert the version back. I'll have to leave Print-Job-History disabled until there's a fix.  I prefer to have the UI plugin over the PrintJobHistory.

I'm posting this issue on both plugin [Issues] board.

The following are log files and System info:
--------------------------------------------------------
From the log files:
Successfully uninstalled Print-Job-History-1.10.0
Successfully installed Print-Job-History-1.12.dev1
Update of PrintJobHistory to 1.12.dev1 successful!
Successfully uninstalled OctoEverywhere-0.7.4
Successfully installed OctoEverywhere-0.8.0
Update of octoeverywhere to 0.8.0 successful!

From OctoPrint System Information: 
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:87.0) Gecko/20100101 Firefox/87.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 3192
env.hardware.ram: 17034551296
env.os.bits: 64
env.os.id: windows
env.os.platform: win32
env.python.pip: 21.0.1
env.python.version: 3.9.0
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin bugfix-2.0.x (Feb 22 2021

From OctoPrint Settings display below [Get More]
"c:\octoprint\venv\scripts\python.exe", Version 21.0.1
Installation directory: c:\octoprint\venv\Lib\site-packages ("--user" flag: no)
Virtual environment: yes
 There are no plugin notices. Great!
Refresh notices from repository 

From About OctorPrint
Version 1.5.3

PS:*** Thanks for this great plugin! ***

David-Maisonave, 2021-04-14:
FYI:
I figured out how to revert PrintJobHistory back to **1.10.0**.
Use the following link to access the zip file.
https://github.com/OllisGit/OctoPrint-PrintJobHistory/tags
Right click the (zip) and copy link location. Then paste it to OctoPrint's plugin manager URL field and click install.
It reverts it even if you haven't uninstalled the new version.

Of course, you should make sure to ignore the message to update plugin to **1.12.dev1**, or any other PrintJobHistory update until this bug is addressed.

LazeMSS, 2021-04-14:
Duplicate of: https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/129

closed 2021-04-14 by LazeMSS