LazeMSS/OctoPrint-UICustomizer #134: [FR] - Blue Moon theme could use a tweak

KenLucke, 2021-04-14:
The dropdown controls in the Blue Moon theme are very hard to read due to color combinations.  A darker text, or a darker background and lighter text, would make it easier on the eyes.

[image]

LazeMSS, 2021-04-14:
Thanks - I will look into it ASAP

LazeMSS, 2021-04-15:
Fixed - restart octoprint to get the update version.

KenLucke, 2021-04-15:
Thank you!  In middle of prints now, will check it out as soon as 1 of the 4 printers frees up (about 5 hours)

KenLucke, 2021-04-15:
Nope, not seein' any difference.  Fired up one of my spare Pi systems, installed UIC, checked Blue Moon, still same colors.

[image]

LazeMSS, 2021-04-15:
Strange will check.. maybe your browser has cached it but that should not the case with a new pi.

KenLucke, 2021-04-15:
That was my thinking.  I refreshed the browser, tried a different browser that had never loaded it before, and finally fired up the other Pi.

LazeMSS, 2021-04-16:
Can you try and open:
http://YOUR_RPI_PI/plugin/uicustomizer/static/themes/css/active.css?theme=bluemoon
YOUR_RPI_PI <- replace with your rpi ip

And search for select {

it should look like this:
```
select {
  width: 220px;
  background-color: #152443;
  border: 1px solid #cccccc;
}
```

KenLucke, 2021-04-16:
Nope:

```
select {
  width: 220px;
  background-color: #cccccc;
  border: 1px solid #bbbbbb;
}
```

LazeMSS, 2021-04-16:
That is really strange - did you allow it to get the remote themes? Its only fixed in that one

KenLucke, 2021-04-16:
> That is really strange - did you allow it to get the remote themes? Its only fixed in that one

I don't see any settings to allow/disallow getting remote theme(s).  They all come up automatically, and I've tried switching between them (and "default") and then back to Blue Moon, with no change.  I've done page reloads, and refreshes, and actually gone into my prefs and cleared out cookie and other data from github, where you have the themes stored.

Is there a place on the local (RPi) drive where they are saved that I can check the saved file version?

KenLucke, 2021-04-16:
Even just tried disabling the plugin, restarting OP, then reenabling it and restarting again.  No change.

KenLucke, 2021-04-16:
Huh,  I just used the URL above and searched, now it's changed, but still not to what you said it should be:

```
select {
  width: 220px;
  background-color: #ffffff;
  border: 1px solid #cccccc;
```

KenLucke, 2021-04-16:
Here's the whole schmere:

```
@import url('//fonts.googleapis.com/css2?family=Roboto:ital,wght@0,100;0,400;1,500&display=swap');

/* LOADER */
#page-container-loading{
  background: #16213e;
}

/* set theme */
#page-container-loading-header:after {
  content: "UI Customizer / Blue moon" !important;
}
h1#page-container-loading-header:before {
  content: "\f6c3";
  font-family: "Font Awesome 5 Free";
  -webkit-font-smoothing: antialiased;
  display: inline-block;
  font-style: normal;
  font-variant: normal;
  text-rendering: auto;
  line-height: 1;
  font-weight: 900;
  position: absolute;
  bottom: 10vh;
  left: 0px;
  width: 100%;
  font-size: 2.5vw;
  animation: pulsate 1s ease-out;
  animation-iteration-count: infinite;
  margin-bottom: -3%;
}

#page-container-loading-header, #page-container-loading-spinner{
  color: #fff !important;
}

/*
 RESET ALL AND MAKE PRELOAD
 */

.icon-black,.octoprint-container .accordion-heading [class*=" icon-"],.octoprint-container .accordion-heading [class^=icon-],.octoprint-container .accordion-heading .accordion-heading-button > a{
  color: inherit !important;
}

/* reset all the really broken styling*/
#navbar .navbar-inner .nav>li.dropdown.open>.dropdown-toggle{
  background-color:  inherit !important;
}

/* fix hardocde links colors*/
.octoprint-container .accordion-heading .accordion-heading-button >a,
table td.gcode_files_action a,table th.gcode_files_action a,
table td.timelapse_files_action a,table td.timelapse_unrendered_action a,table th.timelapse_files_action a,table th.timelapse_unrendered_action a,
table td.settings_groups_actions a,table td.settings_users_actions a,table th.settings_groups_actions a,table th.settings_users_actions a,
table td.settings_printerProfiles_profiles_action a,table th.settings_printerProfiles_profiles_action a{
  color: #7a8288 !important;
  padding-left: 5px;
  padding-right: 5px;
}
.octoprint-container .accordion-heading .accordion-heading-button >a:hover,
table td.gcode_files_action a:hover,table th.gcode_files_action a:hover,
table td.timelapse_files_action a:hover,table td.timelapse_unrendered_action a:hover,table th.timelapse_files_action a:hover,table th.timelapse_unrendered_action a:hover,
table td.settings_groups_actions a:hover,table td.settings_users_actions a:hover,table th.settings_groups_actions a:hover,table th.settings_users_actions a:hover,
table td.settings_printerProfiles_profiles_action a:hover,table th.settings_printerProfiles_profiles_action a:hover{
  color: #3a87ad !important;
}

/* set to muted color*/
[... 6525 lines cut ...]
#files .gcode_files .entry:hover{
  background-color: #3a87ad !important;
  border-bottom: 1px solid #46a1ce;
}

#files .gcode_files .entry:hover > * {
  color: #fff;
}

#files .gcode_files .entry div[style="font-weight: bold;"]{
  color: #fff;
  font-weight: 500 !important;
}

#files .gcode_files .entry:hover div[style="font-weight: bold;"]{
  color: #fff;
  font-weight:500 !important;
}

.UICHeaderIcons .navbar-text{
  font-weight: normal;
  padding-right: 15px;
  padding-left: 15px;
}
```

LazeMSS, 2021-04-16:
What version of the plugin are you running?

KenLucke, 2021-04-16:
[image]

LazeMSS, 2021-04-16:
Something broke?

KenLucke, 2021-04-16:
I cleared all the logs, then restarted OP and changed themes twice (from the initially loaded one). (notably, I turned on the octoprint.plugins.uicustomizer log, both in DEBUG and INFO modes, but no logs ever appeared)

Here are all the references I pulled from the octoprint.log, using a search grep of: `^.*\r^.*uicustomizer.*\r.*\r.*.\r` to give a couple of surrounding lines:

```
2021-04-16 13:38:18,920 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for z_probe_offset from AST of /home/pi/oprint/lib/python3.7/site-packages/octoprint_z_probe_offset/__init__.py
2021-04-16 13:38:18,936 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for uicustomizer from AST of /home/pi/oprint/lib/python3.7/site-packages/octoprint_uicustomizer/__init__.py
2021-04-16 13:38:18,946 - octoprint.plugin.core - DEBUG - Parsing plugin metadata for thespaghettidetective from AST of /home/pi/oprint/lib/python3.7/site-packages/octoprint_thespaghettidetective/__init__.py
2021-04-16 13:38:18,956 - octoprint.plugin.core - INFO - Plugin Access Anywhere - The Spaghetti Detective (1.5.1) is disabled.

2021-04-16 13:38:19,724 - octoprint.plugin.core - DEBUG - Loaded plugin z_probe_offset: Z Probe Offset (0.5)
2021-04-16 13:38:19,724 - octoprint.plugin.core - DEBUG - Loaded plugin uicustomizer: UI Customizer (0.1.5.6)
2021-04-16 13:38:19,725 - octoprint.plugin.core - DEBUG - Loaded plugin terminalcommandsextended: Terminal Commands Extended (0.1.6)
2021-04-16 13:38:19,725 - octoprint.plugin.core - DEBUG - Loaded plugin TemperatureLegendMover: TemperatureLegendMover (1.0.0)

2021-04-16 13:38:19,878 - octoprint.plugin.core - DEBUG - Enabled plugin z_probe_offset: Z Probe Offset (0.5)
2021-04-16 13:38:19,881 - octoprint.plugin.core - DEBUG - Enabled plugin uicustomizer: UI Customizer (0.1.5.6)
2021-04-16 13:38:19,885 - octoprint.plugin.core - DEBUG - Enabled plugin terminalcommandsextended: Terminal Commands Extended (0.1.6)
2021-04-16 13:38:19,889 - octoprint.plugin.core - DEBUG - Enabled plugin TemperatureLegendMover: TemperatureLegendMover (1.0.0)

2021-04-16 13:38:23,731 - octoprint.plugin.core - DEBUG - Initialized plugin mixin implementation for plugin z_probe_offset
2021-04-16 13:38:23,746 - octoprint.plugin.core - DEBUG - Initialized plugin mixin implementation for plugin uicustomizer
2021-04-16 13:38:23,749 - octoprint.plugin.core - DEBUG - Initialized plugin mixin implementation for plugin terminalcommandsextended
2021-04-16 13:38:23,752 - octoprint.plugin.core - DEBUG - Initialized plugin mixin implementation for plugin TemperatureLegendMover

|  Translate Model Plugin (0.1.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_translatemodel
|  UI Customizer (0.1.5.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_uicustomizer
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
|  Z Probe Offset (0.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_z_probe_offset

2021-04-16 13:38:24,653 - octoprint.server - DEBUG - Registered assets of plugin translatemodel under URL prefix /plugin/translatemodel
2021-04-16 13:38:24,658 - octoprint.server - DEBUG - Registered assets of plugin uicustomizer under URL prefix /plugin/uicustomizer
2021-04-16 13:38:24,663 - octoprint.server - DEBUG - Registered assets of plugin z_probe_offset under URL prefix /plugin/z_probe_offset
2021-04-16 13:38:25,063 - octoprint.server - DEBUG - Adding additional route /plugin/arc_welder/downloadFile handled by handler <class 'octoprint_arc_welder.ArcWelderLargeResponseHandler'> and with additional arguments {'request_callback': <bound method ArcWelderPlugin.download_file_request of <octoprint_arc_welder.ArcWelderPlugin object at 0x7f863fcb00>>, 'as_attachment': True, 'access_validation': <function validation_chain.<locals>.f at 0x7f8401cf28>}

2021-04-16 13:38:27,915 - octoprint.plugin - DEBUG - Calling on_startup on translatemodel
2021-04-16 13:38:27,915 - octoprint.plugin - DEBUG - Calling on_startup on uicustomizer
2021-04-16 13:38:27,920 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2021-04-16 13:38:27,923 - octoprint.plugin - DEBUG - Calling on_after_startup on announcements

2021-04-16 13:38:32,624 - octoprint.plugin - DEBUG - Calling on_after_startup on translatemodel
2021-04-16 13:38:32,668 - octoprint.plugin - DEBUG - Calling on_after_startup on uicustomizer
2021-04-16 13:38:32,668 - octoprint.plugins.uicustomizer - INFO - UI Customizer is initialized.
2021-04-16 13:38:33,222 - octoprint.server.util.sockjs - DEBUG - Socket message held back until permissions cleared, added to backlog: plugin

2021-04-16 13:39:10,450 - octoprint.server.util.flask - DEBUG - Serving entry for / from cache (key: ui:_default:https://192.168.1.50/:en)
2021-04-16 13:39:28,705 - octoprint.plugins.uicustomizer - INFO - Setting theme "cyborg"
2021-04-16 13:39:31,166 - octoprint.events.fire - DEBUG - Firing event: SettingsUpdated (Payload: {'config_hash': '6ee8b87ab91fdf3a9fb580eff053ccfe', 'effective_hash': 'e35f38df190d06558bd5197eb08e2862'})
2021-04-16 13:39:31,167 - octoprint.events - DEBUG - Sending action to <function Server.run.<locals>.on_settings_update at 0x7f8d162400>

2021-04-16 13:39:36,812 - octoprint.server.util.flask - DEBUG - Serving entry for / from cache (key: ui:_default:https://192.168.1.50/:en)
2021-04-16 13:39:52,806 - octoprint.plugins.uicustomizer - INFO - Setting theme "bluemoon"
[... cut ...]
2021-04-16 13:40:00,719 - octoprint.server.util.flask - DEBUG - Serving entry for / from cache (key: ui:_default:https://192.168.1.50/:en)
2021-04-16 13:40:13,156 - octoprint.plugins.uicustomizer - INFO - Setting theme "default"
2021-04-16 13:40:15,670 - octoprint.events.fire - DEBUG - Firing event: SettingsUpdated (Payload: {'config_hash': 'a57ba80052269e9e9bc1e0da6be9c5d4', 'effective_hash': 'fe31518ea30f7e15c510ce0b327e19bc'})
2021-04-16 13:40:15,671 - octoprint.events - DEBUG - Sending action to <function Server.run.<locals>.on_settings_update at 0x7f8d162400>
```

LazeMSS, 2021-04-16:
It seems like your plugin is stuck on local version and never loading the external themes.

1. What does your config (/home/pi/.octoprint/config.yaml) say regarding themeLocal ie:

```
 uicustomizer:
    rows:
    - _00#connection_wrapper: true
      _01#files_wrapper: true
      _02#sidebar_plugin_firmware_check_warning_wrapper: false
      _03#sidebar_plugin_firmware_check_info_wrapper: false
    - _00div.UICmainTabs: true
    - _00#UICGcodeVWidget: true
      _01#UICWebCamWidget: true
      _02#sidebar_plugin_action_command_notification_wrapper: true
      _03#state_wrapper: true
    theme: slate
    **themeLocal: false**
    themeVersion: 0.1.4
    topIconSort:
    - navbar_plugin_psucontrol
    - navbar_plugin_action_command_prompt
    - navbar_plugin_toptemp
    - navbar_plugin_logging_seriallog
    - navbar_plugin_logging_plugintimingslog
    - navbar_plugin_pi_support
    widths:
    - '3'
    - '6'
    - '3'

```

2. Do you have the local storage "plugin.uicustomizer.getThemesApproved" set to what?

KenLucke, 2021-04-16:
1)

 ```
    uicustomizer:
    compactMenu: false
    filesFullHeight: false
    hideMainCam: true
    mainTabs:
    - - tab_plugin_dashboard_link
      - true
      - false
      - fas fa-tachometer-alt
      - iconOnly
      - false
    - - control_link
      - true
      - false
      - fas fa-gamepad
      - iconOnly
      - false
    - - gcode_link
      - true
      - false
      - fab fa-codepen
      - iconOnly
      - false
    - - temp_link
      - true
      - false
      - fas fa-chart-line
      - iconOnly
      - false
    - - term_link
      - true
      - false
      - fas fa-terminal
      - iconOnly
      - false
    - - tab_plugin_bedlevelvisualizer_link
      - true
      - false
      - fas fa-procedures
      - iconOnly
      - false
    - - tab_plugin_cancelobject_link
      - true
      - false
      - fas fa-ban
      - iconOnly
      - false
    - - tab_plugin_multi_colors_link
      - true
      - false
      - fas fa-palette
      - iconOnly
      - false
    - - tab_plugin_filemanager_link
      - true
      - false
      - fas fa-folder
      - iconOnly
      - false
    - - tab_plugin_eeprom_marlin_link
      - true
      - false
      - far fa-list-alt
      - iconOnly
      - false
    - - tab_plugin_arc_welder_link
      - true
      - false
      - fas fa-circle-notch
      - iconOnly
      - false
    - - tab_plugin_translatemodel_link
      - true
      - false
      - fas fa-arrows-alt
      - iconOnly
      - false
    rows:
    - _00#state_wrapper: true
      _01#files_wrapper: true
      _02#connection_wrapper: true
      _03#sidebar_plugin_firmware_check_warning_wrapper: false
      _04#sidebar_plugin_firmware_check_info_wrapper: false
    - _00div.UICmainTabs: true
    - _00#sidebar_plugin_macro_wrapper: true
      _01#UICGcodeVWidget: true
      _02#UICWebCamWidget: true
      _03#sidebar_plugin_action_command_notification_wrapper: false
    theme: slate
    topIconSort:
    - navbar_plugin_DisplayLayerProgress
    - navbar_plugin_cancelobject
    - navbar_plugin_firmwareupdater
    - navbar_plugin_display_eta
    - navbar_plugin_action_command_prompt
    - navbar_plugin_simpleemergencystop
    - navbar_plugin_logging_seriallog
    - navbar_plugin_logging_plugintimingslog
    - navbar_plugin_pi_support
    - navbar_plugin_toptemp
    widths:
    - '3'
    - '6'
    - '3'
```

2)   Where is this setting? I have not seen it
[image]

3) Bingo.  It was showing some blocked elements.  Tracing it down, I have NoScript installed.  `github.com` is whitelisted,  but `github.io` was NOT.

[image]

KenLucke, 2021-04-16:
`themelocal: false` now shows up in config,.yaml, too.

LazeMSS, 2021-04-16:
AHHH thanks I was looking for that
1. Looks fine
2. Its not a setting - its a local storage thing
3. Awesome

Thanks for all the help - I will add a bigger warning when failing to download externally themes.

Closing now ;)

closed 2021-04-16 by LazeMSS

KenLucke, 2021-04-16:
> Thanks for all the help - I will add a bigger warning when failing to download externally themes.

Actually, it is I that should be thanking YOU.