LazeMSS/OctoPrint-UICustomizer #137: Not an issue, info request

KenLucke, 2021-04-14:
Is there somewhere that I can get information on making/customizing themes?  The CSS identifiers for the Custom CSS box and such.  I'd like to make my own themes, and eliminate Themeify if I can do so.

LazeMSS, 2021-04-14:
All themes are loaded from here:

https://github.com/LazeMSS/OctoPrint-UICustomizerThemes 

Take a look at the CSS files in there ;)

closed 2021-04-14 by LazeMSS

KenLucke, 2021-04-14:
OK, so how do I 
a) either load a modified copy of one of those (preferable to me), or
b) use the custom CSS box.    I'm not real CSS savvy anymore, I lost track several versions back :)|

For instance, if I want to change the `red` and `red-transparent` colors to my preferred RGB or Hex values for corresponding Yellow or Blue in the Red Night theme, which has the following VARs, how would I format the overriding  CSS (I know I'd have to leave it *named* red if I'm just overriding, otherwise I'd have to change every place the VAR is used):

```
:root {
    --accent: #d32f2f;
    --accent-transparent: rgba(211, 47, 47, .45);
    --accent-darker: #ab2424;
    --background: #1e1e20;
    --background-darker: #121213;
    --grey1: #19191b;
    --grey2: #242424;
    --grey3: #2d2d2f;
    --grey3-dark: #212122;
    --grey4: #3e3e3e;
    --red: #f44336;
    --red-transparent: rgba(244, 67, 54, .7);
    --green: #4caf50;
    --quiteWhite: #f7f7f7;
    --quiteWhite-transparent: hsla(0, 0%, 97%, .3);
    --quiteWhite-dark: #dedede;
    --white: #fff;
    --white-dark: #f2f2f2;
    --grey: #999;
    --black-transparent: hsla(0, 0%, 7%, .6);
    --icon-quiteWhite-Filter: hue-rotate(0deg) saturate(0) brightness(96.8627451%);
}
```

LazeMSS, 2021-04-14:
Ahh I thought you were aiming for creating new themes and submitting them.

I think it will work if you under advanced/custom css just add:
```
:root {
--red: #2340EB;
--red-transparent: rgba(35, 64, 235, .7);
}
```

I the next version themes will have different color schemes to pick from - so the red night theme might just be "night" theme and you can pick red, blue etc.

KenLucke, 2021-04-14:
Tried that.   Didn't work.  :(   Tried using Preview, actually Saving, and Reloading OP, and even went as far as Restarting OP :)

>I the next version themes will have different color schemes to pick from - so the red night theme might just be "night" theme and you can pick red, blue etc.

Cool.  I'll take one of each in Red, Green, Yellow, & Blue :)  For now  :)
(my 4 printers are color coordinated/named so I can tell them apart at a glance, which is why I like different color schemes for each of my OctoPrint instances - Themeify let me do that, but it was waaay too complicated to make sure I got all the little details changed to the right color)

LazeMSS, 2021-04-14:
Strange - I will look into why that didn't work.

And yes the color schemes are comming together with another new feature ;)