LazeMSS/OctoPrint-UICustomizer #141: [Bug] "Hide Main Webcam" does not work

KenLucke, 2021-04-15:
Enabling the "Hide Main Webcam" option does not hide the webcam - I still get it on all tabs (which appears to slow things down a bit, the camera feed gets stuttery).  

I use Dashboard, because I like the diagnostic information it displays, and I like having the webcam only on the Dashboard tab.   I tried turning all the others off with this option, and they persist, even after a restart.  I tried disabling Dashboard to see if it was somehow causing the option to be ignored, but nothing changed.    I also usually run a webcam feed only, in another web browser tab, if I want to really see what's happening larger size, and more smoothly.

It'd be nice to have the choice to enable or disable the webcam from each tab, too... I don't need a webcam showing all the time I'm doing everything, such as working in the Bed Visualizer or File Manager tabs .... or make a mini version (such as the mini version of the gcode viewer) that can be enabled in the sidebar.

Sorry to be a pain :)  I've beta tested stuff since the first Apple (not even Apple II) days in the 80s, and I'm always pushing the envelope :)

LazeMSS, 2021-04-15:
No need to excuse. I love feedback.

KenLucke, 2021-04-15:
> No need to excuse. I love feedback.

Don't encourage me :)

LazeMSS, 2021-04-16:
Please split issues and feature request.
Regarding the Hide main webcam not working - I'm not sure what you mean. 

"Hide main webcam" hides the webcam from the Control tab.
There is a mini webcam similar to the gcode as you say  - so if you are talking about the webcam as a widget (right hand side af default) then just go the "Layout" tab inside UI Customizer and click the eye (hidden when there is a slash over it)

[image]

[image]

LazeMSS, 2021-04-16:
@KenLucke ?

KenLucke, 2021-04-16:
Yes?

LazeMSS, 2021-04-16:
Can you try and explain a bit more - im not sure what the problem is?

KenLucke, 2021-04-16:
> "Hide main webcam" hides the webcam from the Control tab and nothing else

This explains a lot.  I had the webcam set to the center panel, and it was shoing the Dashboard webcam PLUS the UIC webcam, both on the dashboard page. I was trying to turn the UIC one off, as it was not needed...the Dashboard version, plus my separate tab full window versuion for close inspection.  Moving the camera off to the sidebar and turning off the eye solved my problem.

LazeMSS, 2021-04-16:
Okay I was reading it again and again and I couldn't make sense of it :)

I will rename that option :)

Okay to close?

KenLucke, 2021-04-16:
> Okay to close?

Fine by me.

closed 2021-04-16 by LazeMSS