LazeMSS/OctoPrint-UICustomizer #142: [Bug] Conflict with Gcode viewer tab/Cancel Objects

KenLucke, 2021-04-15:
Sorry to hit you with another one.  I started this bug report over there, but it seems to be interaction between the two.

https://github.com/paukstelis/Octoprint-Cancelobject/issues/67

LazeMSS, 2021-04-16:
Can you provide med with a sample file to test?

KenLucke, 2021-04-16:
This is the one I've been using as an example.

https://www.dropbox.com/s/vcc5ycrkzmx8ybi/Root_Ball_Mould%20x6%20%5BPLA%5D.aw.gcode.zip

LazeMSS, 2021-04-16:
What browser are you using - on the other https://github.com/paukstelis/Octoprint-Cancelobject/issues/67 issue it seems like having the fullsize gcode doesn't really work.

I really can't figure out why this is not working I have support for the Cancelobject plugin.
But on the other hand I can really figure out how to get the markers showing on my virtual printer - there are no orange dots showing up no matter what I do so please guide me ;)

KenLucke, 2021-04-16:
I've tried both Firefox (Mac) and Safari.  Same results.

I don't think the markers show on the virtual printer.  I seem to recall (maybe) reading something about that a long time ago, but I may be mistaken.

LazeMSS, 2021-04-16:
Okay my real printer is broken at the moment.
Do the markers show up when loading the file or do I need to start printing the file?

KenLucke, 2021-04-16:
they only show up after the first layer is finished (actually, as each object is finished for the first time).

LazeMSS, 2021-04-16:
Okay thanks - I will try and test more.

KenLucke, 2021-04-16:
You sometimes also need to do a "Refresh Objects"

LazeMSS, 2021-04-23:
There is no easy fix - will remove support for it when Cancel Object is installed.

LazeMSS, 2021-04-25:
"Fixed" in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.7

Please reopen/comment if not working as expected.

closed 2021-04-25 by LazeMSS