LazeMSS/OctoPrint-UICustomizer #143: 0.1.5.6 doesn't support SpoolManger Window

diedder, 2021-04-18:
together with SpoolManager Plugin (1.3.4) the popup window for editing spools isn't shown completely. Some mandatory fields (such as weight) are not visible so the plugin could not be used.

see screenshots with enabled and disabled UI customizer attached
[image]
[image]

thanks in advance!

LazeMSS, 2021-04-18:
No really the spool manager is pushing the envelope on how much stuff you can cram into a overlay dialog and when bootstrap default responsive CSS is added by UI customizer it breaks. But I think it is fixed in some of the latest versions of spool manager.

I'm hoping at some point to help with a rewrite of the UI for spool manager.

Btw what screen resolution are you using?

diedder, 2021-04-18:
my screen on laptop is MBP 13,3-Zoll (2560 × 1600)

LazeMSS, 2021-04-18:
Strange that should be plenty screen estate. I don't have any problems. Will test again

LazeMSS, 2021-04-23:
@diedder I can't recreate - can you try again and maybe clear browser cache.

LazeMSS, 2021-04-23:
Found the bug - https://github.com/OllisGit/OctoPrint-SpoolManager/issues/142#issuecomment-825601100

I won't fix - should be fixed in the SpoolManager plugin

closed 2021-04-23 by LazeMSS

diedder, 2021-04-23:
Thank you anyway!

LazeMSS, 2021-04-25:
I have included a fix in the next release but i hope the main author will add this as well

lightmaster, 2021-04-25:
@LazeMSS You sir are a hero. 👏