LazeMSS/OctoPrint-UICustomizer #144: Add support for OctoPrint-CameraSettings

LazeMSS, 2021-04-22:
https://github.com/The-EG/OctoPrint-CameraSettings/issues/6

LazeMSS, 2021-04-25:
Done but not perfect - the flexbox stuff realy don't mix good with the old boostrap responsive way

The-EG, 2021-04-25:
> Done but not perfect - the flexbox stuff realy don't mix good with the old boostrap responsive way

Thank you! If there's a 'better' way that would be more straight forward or appropriate for this and the default UI I'd certainly be open to feedback and learning a different way.

LazeMSS, 2021-04-25:
"Fixed" in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.7
Please reopen/comment if not working as expected.

@The-EG  I will make a proper PR for my ideas for the settings :)

closed 2021-04-25 by LazeMSS