LazeMSS/OctoPrint-UICustomizer #146: Unable to change favicon

krudoy, 2021-04-23:
## Description
Change of color in 'Octoprint' -> 'Appearance' does not change color of favicon when the plugin is enabled. Even more - it resets the color value to default :(

**Plugin Disabled:**, Favicon is orange
[image]

**Plugin Enabled:**, Favicon is green
[image]

<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.5.6**
- Top Temp[toptemp] v0.0.1.4
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.4
- Preheat Button[preheat] v0.7.2
- Cura Thumbnails[UltimakerFormatPackage] v0.2.2
- Translate Model Plugin[translatemodel] v0.2.0
- Resource Monitor[resource_monitor] v0.3.3
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- PrettyGCode[prettygcode] v1.2.4
- OctoText[OctoText] v0.2.0
- MultiCam[multicam] v0.2.8
- InlineConfirm[inlineconfirm] v1.1.0
- GcodeEditor[GcodeEditor] v0.2.12
- Fullscreen Plugin[fullscreen] v0.0.6
- FileManager[filemanager] v0.1.6
- Fan Speed Control[fanspeedslider] v0.2.1
- Exclude Region[excluderegion] v0.3.0
- Octoprint-Display-ETA[display_eta] v1.1.3
- Dashboard[dashboard] v1.18.1
- M150 Control[M150Control] v0.2.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.26.0
- Bed Visualizer[bedlevelvisualizer] v1.0.1
- Backup Scheduler[backupscheduler] v0.0.5
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: false

**Software versions**
- OctoPrint 1.5.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.72 Safari/537.36

LazeMSS, 2021-04-24:
Yes in the next release this will just show a warning I was just tired of people complaining about broken themes because they had turned on colors or transparency 🙈

LazeMSS, 2021-04-24:
Closing this, duplicate of https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/135

PS I'm working on adding something similar to the plugin allowing you to set the favicon

closed 2021-04-24 by LazeMSS