LazeMSS/OctoPrint-UICustomizer #148: 0.1.5.7 stopped to work

Guilouz, 2021-04-25:
Hi,

Just updating to 0.1.5.7 and UICustomizer stopped to work, after reboot plugin has disappeared from my plugin list.

Try to reinstall it manually, same issue.

I'm on OctoPrint 1.6.0rc3 release.

scaramonga, 2021-04-25:
Yup, same thing for me.  Just updated just now, and its no longer showing in system. 1.6.0rc3 also.

LazeMSS, 2021-04-25:
Anything in the log files etc?

LazeMSS, 2021-04-25:
im not on 1.6.0RC3 yet

Guilouz, 2021-04-25:
Plugin is detected as unkwown name. No issue when re-installing 0.1.5.6 over 0.1.5.7

LazeMSS, 2021-04-25:
Bug found - fix on its way

scaramonga, 2021-04-25:
2021-04-25 19:37:28,329 - octoprint.plugins.softwareupdate - INFO - Starting update of uicustomizer to 0.1.5.7...
2021-04-25 19:37:34,024 - octoprint.plugins.softwareupdate - INFO - Update of uicustomizer to 0.1.5.7 successful!
2021-04-25 19:37:34,165 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2021-04-25 19:37:34,166 - octoprint.plugins.softwareupdate - INFO - Restarting...
2021-04-25 19:37:34,237 - octoprint.server - INFO - Shutting down...
2021-04-25 19:37:37,781 - octoprint.plugin.core - ERROR - Error loading plugin uicustomizer

LazeMSS, 2021-04-25:
Fixed now - please try the new one now.

bjoernbusch, 2021-04-25:
works like a charm! thanks!

closed 2021-04-25 by LazeMSS