LazeMSS/OctoPrint-UICustomizer #150: Text in white area almost unreadable

ipa64, 2021-04-29:
Beautiful but text in input field with white background are almost unreadable 
Example terminal input field
How to change that ?
Thank you

LazeMSS, 2021-04-29:
Which theme?

ipa64, 2021-04-29:
Red night ( but I tried Dark green too, samething)

LazeMSS, 2021-04-30:
I can't recreate - please post a screenshot and report a bit more info.
browser
plugins installed

ipa64, 2021-04-30:
Safari 14.0.3
Here some print screen

[https://i84.servimg.com/u/f84/20/23/21/77/captur10.png](url)
[https://i84.servimg.com/u/f84/20/23/21/77/captur11.png](url)
[https://i84.servimg.com/u/f84/20/23/21/77/captur12.png](url)

Thank you

ipa64, 2021-04-30:
I have just understand what was the problem !
I have Themify installed too and I thought it was disable but not !
So now all is ok, sorry to have bothered you
Thank you

LazeMSS, 2021-04-30:
Okay thanks

closed 2021-04-30 by LazeMSS