LazeMSS/OctoPrint-UICustomizer #151: Cosmetic

KenLucke, 2021-05-01:
When "Fixed Footer/Bottom Bar" is active, it cuts off the very bottom of the scrollable area.

[image]
[image]
[image]

LazeMSS, 2021-05-02:
What screensize?

LazeMSS, 2021-05-02:
Do you maybe have another plugin installed?

KenLucke, 2021-05-02:
I've tried several different screen resolutions, and it happens on all of them.  And yes, I have many other plugins installed.

LazeMSS, 2021-05-02:
> I've tried several different screen resolutions, and it happens on all of them. And yes, I have many other plugins installed.

Any plugins that might change the bottom bar etc?
And does it happen with all themes?

LazeMSS, 2021-05-02:
Give me a guide on how to recreate it and I will fix it :)

KenLucke, 2021-05-02:
> > I've tried several different screen resolutions, and it happens on all of them. And yes, I have many other plugins installed.
> 
> Any plugins that might change the bottom bar etc?
> And does it happen with all themes?

Nothing that affects that in my knowledge.  It happens with at least two of them, I have not checked the others.

I don't do anything special to recreate it.  Just enable the bottom bar in UIC, and find a tab or two that require scrolling to get all the way to the bottom, then scroll all the way down.

KenLucke, 2021-05-02:
Yes, it happens with all themes, I just ran through them (all I can do at the moment, I have every printer running, so I can't enable/disable plugins or try anything that would require a restart of OP.

LazeMSS, 2021-05-02:
Ohh a tab is causing this.... Will look

KenLucke, 2021-05-02:
[image]

KenLucke, 2021-05-02:
[image]

KenLucke, 2021-05-02:
Basically, I think all it needs is a CSS tweak to make the main window area [BottomBar height less] when the bottom bar is on than it is when it is off.

LazeMSS, 2021-05-03:
Okay I will test in Firefox now - give me a day or two - kinda frantic at the moment

KenLucke, 2021-05-03:
No worries, it's just cosmetic at the worst (of course, your entire plugin is cosmetic, but what the heck < grin > )

LazeMSS, 2021-05-08:
@KenLucke spent good 30 min. on this one I can't recreate it - either some custom css or another plugin is messing with something - no matter what i try i can't recreate it. I would really like to fix it so maybe you could try and list your plugins.

KenLucke, 2021-05-08:
[plugin_list.json.zip](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/6446642/plugin_list.json.zip)

LazeMSS, 2021-05-15:
@KenLucke  I still haven't figured out whats causing this...

I haven't tried all the plugins listed but I noticed a scroll to the "top arrow" on you screenshots - how did you get that? I'm missing that and that might be causing it.

KenLucke, 2021-05-15:
That is from the Floating Navbar plugin.

LazeMSS, 2021-05-17:
I can't recreate it - I thought it might be due to all the custom buttons below the terminal but doesn't seem to be the case.

1. Does it happen when using the default theme?
2. What is the pixel height of the footer when fixed is on: `$('.footer.navbar.navbar-fixed-bottom').height();`

LazeMSS, 2021-05-25:
@KenLucke any news?

LazeMSS, 2021-06-09:
@KenLucke  ping?

LazeMSS, 2021-06-28:
Closing - haven't heard from @KenLucke  in 19 days - can't recreate :/

closed 2021-06-28 by LazeMSS