LazeMSS/OctoPrint-UICustomizer #152: How do you add custom themes?

NovaViper, 2021-05-05:
Hey! I'm wondering how can I add custom themes? I made one based off the [Dracula theme suite](https://github.com/dracula/dracula-theme) and I want to upload it to see how it looks.

This is the theme css for it, I took the red_night css and just changed the colors for the Dracula theme (or at least to the best of my ability); if you find something wrong with it you can change it!
[dracula.zip](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/6425037/dracula.zip)

LazeMSS, 2021-05-05:
here: https://github.com/LazeMSS/OctoPrint-UICustomizerThemes - do a pull request (PR) if possible.

If not then I can try and add the your theme but i prefer PR

KenLucke, 2021-05-05:
On a related question, where does UIC store the local copy of the themes on the Pi?  I'd like to modify one or two (just color changes) before you get around to implementing your idea you had in another thread.  I've dug around but can't seem to find them.

LazeMSS, 2021-05-06:
> On a related question, where does UIC store the local copy of the themes on the Pi? I'd like to modify one or two (just color changes) before you get around to implementing your idea you had in another thread. I've dug around but can't seem to find them.

You can use custom CSS under advanced. The actual files will be overwritten by updates etc

KenLucke, 2021-05-06:
We tried that, it didn't work :(

NovaViper, 2021-05-07:
> We tried that, it didn't work :(

Did you put the entire file's contents into the custom css page? I did that and it loaded up the elements :D
[image]

KenLucke, 2021-05-07:
No, I didn't try that.  I just tried to override the root colors, even using !important to try to force them.  I'll give your method a try.

LazeMSS, 2021-05-08:
@KenLucke  and @NovaViper you can overwrite the color variables like below (take from red night) with custom css - you must include the root part - and also notice the custom css preview only updates on "blur" of the input field:
```
:root {
    --accent: #d32f2f;
    --accent-transparent: rgba(211, 47, 47, .45);
    --accent-darker: #ab2424;
    --background: #1e1e20;
    --background-darker: #121213;
    --grey1: #19191b;
    --grey2: #242424;
    --grey3: #2d2d2f;
    --grey3-dark: #212122;
    --grey4: #3e3e3e;
    --red: #f44336;
    --red-transparent: rgba(244, 67, 54, .7);
    --green: #4caf50;
    --quiteWhite: #f7f7f7;
    --quiteWhite-transparent: hsla(0, 0%, 97%, .3);
    --quiteWhite-dark: #dedede;
    --white: #fff;
    --white-dark: #f2f2f2;
    --grey: #999;
    --black-transparent: hsla(0, 0%, 7%, .6);
    --icon-quiteWhite-Filter: hue-rotate(0deg) saturate(0) brightness(96.8627451%);
}

```

KenLucke, 2021-05-08:
I tried this on @NovaViper's suggestion .  It worked until I reloaded OP, then it showed, but instantly reverted back to the original theme colors.

APEbbers, 2021-05-13:
Hi

If you want override a specific style with CustomCSS, you can use " !Important" at the end of each line. I've used this to restore the colors for macro buttons from the "sidebar macros" plugin.  These styles where overridden when using the red night theme.

See below the custom css that i've used:
.btn-info {
    color: #fff !Important;
    text-shadow: 0 -1px 0 rgb(0 0 0 / 25%) !Important;
    background-color: #49afcd !Important;
    background-image: -moz-linear-gradient(top,#5bc0de,#2f96b4) !Important;
    background-image: -webkit-gradient(linear,0 0,0 100%,from(#5bc0de),to(#2f96b4)) !Important;
    background-image: -webkit-linear-gradient(top,#5bc0de,#2f96b4) !Important;
    background-image: -o-linear-gradient(top,#5bc0de,#2f96b4) !Important;
    background-image: linear-gradient(to bottom,#5bc0de,#2f96b4) !Important;
[image]

    background-repeat: repeat-x !Important;
    border-color: #2f96b4 #2f96b4 #1f6377 !Important;
    border-color: rgba(0,0,0,.1) rgba(0,0,0,.1) rgba(0,0,0,.25) !Important;
    *background-color: #2f96b4 !Important;
    filter: progid:DXImageTransform.Microsoft.gradient(enabled=false) !Important;
} 

.btn-warning {
    color: #fff;
    text-shadow: 0 -1px 0 rgb(0 0 0 / 25%) !Important;
    background-color: #faa732 !Important;
    background-image: -moz-linear-gradient(top,#fbb450,#f89406) !Important;
    background-image: -webkit-gradient(linear,0 0,0 100%,from(#fbb450),to(#f89406)) !Important;
    background-image: -webkit-linear-gradient(top,#fbb450,#f89406) !Important;
    background-image: -o-linear-gradient(top,#fbb450,#f89406) !Important;
    background-image: linear-gradient(to bottom,#fbb450,#f89406) !Important;
    background-repeat: repeat-x !Important;
    border-color: #f89406 #f89406 #ad6704 !Important;
    border-color: rgba(0,0,0,.1) rgba(0,0,0,.1) rgba(0,0,0,.25) !Important;
    *background-color: #f89406 !Important;
    filter: progid:DXImageTransform.Microsoft.gradient(enabled=false) !Important;
}

.btn-danger {
    color: #fff !Important;
    text-shadow: 0 -1px 0 rgb(0 0 0 / 25%) !Important;
    background-color: #da4f49 !Important;
    background-image: -moz-linear-gradient(top,#ee5f5b,#bd362f) !Important;
    background-image: -webkit-gradient(linear,0 0,0 100%,from(#ee5f5b),to(#bd362f)) !Important;
    background-image: -webkit-linear-gradient(top,#ee5f5b,#bd362f) !Important;
    background-image: -o-linear-gradient(top,#ee5f5b,#bd362f) !Important;
    background-image: linear-gradient(to bottom,#ee5f5b,#bd362f) !Important;
    background-repeat: repeat-x !Important;
    border-color: #bd362f #bd362f #802420 !Important;
    border-color: rgba(0,0,0,.1) rgba(0,0,0,.1) rgba(0,0,0,.25) !Important;
    *background-color: #bd362f !Important;
    filter: progid:DXImageTransform.Microsoft.gradient(enabled=false) !Important;
}

[image]

HugoCLSC, 2021-07-02:
Hi newbi here. I don't know much but wanted to create a custom UI and was using this plugin as a base. 
I was hoping i could get some help on how you "erased" the default UI and inserted yours. 
Did you do that using the javascript file?
Do you recommend usigng the mixin template from octoprint to create a new template and then customize the UI ?
Thanks in advance!

LazeMSS, 2021-07-02:
> Hi newbi here. I don't know much but wanted to create a custom UI and was using this plugin as a base.
> I was hoping i could get some help on how you "erased" the default UI and inserted yours.
> Did you do that using the uicustomizer.js file?
> Do you recommend usigng the mixin template from octoprint to create a new template and then customize the UI ?
> Thanks in advance!

I built JavaScript to remove the hardcoded CSS etc.

If you want to create a completely new UI there is examples of that in the official documentation.

If you want to create a new theme I would recommend looking into the themes in my plugin as a base

HugoCLSC, 2021-07-02:
Thanks for the reply! i'll dig into that a bit more, seems like i missed some documentation.

darthtate, 2022-12-30:
How can you modify the following css in the advanced custom css tab?

#page-container-loading{
  background: linear-gradient(180deg, rgba(6,6,6,1) 0%, rgb(21 77 105) 100%);
}

/* set theme */
#page-container-loading-header:after {
  content: "UI Customizer / Cyborg" !important;
}
h1#page-container-loading-header:before {
  content: "\f544";
  font-family: "Font Awesome 5 Free";
  -webkit-font-smoothing: antialiased;
  display: inline-block;
  font-style: normal;
  font-variant: normal;
  text-rendering: auto;
  line-height: 1;
  font-weight: 900;
  position: absolute;
  bottom: 10vh;
  left: 0px;
  width: 100%;
  font-size: 2.5vw;
  animation: pulsate 1s ease-out;
  animation-iteration-count: infinite;
  margin-bottom: -3%;
}

I am trying to make changes to the boot screen. I want to make changes to the default \f544 icon and also the content: "UI Customizer / Cyborg" areas. Can this be done in this advanced tab or not? I have figured out how to make changes to the default cyborg template to change colors etc, but want to change the boot screen.

LazeMSS, 2022-12-30:
No sadly the custom css stuff is loaded after the loading screen - i have been consindering make it smarter so its injected into the main theme file - I will try and test that because the way the contect is being loaded has been changed in the next release

darthtate, 2022-12-31:
No worries. Thanks for an awesome plugin!