LazeMSS/OctoPrint-UICustomizer #154: Whenever In change my theme's the plug-in backgrounds come in white with grey text

Pen16, 2021-05-10:
## Description
Whenever In change my theme's the plug-in backgrounds come in white with grey text which makes it very hard to read and def makes the theme a lot less appealing. I tried looking into the source code to maybe change the background colors myself but I honestly don't know what I'm doing. I have tried disabling plugins that maybe had anything to do with UI changes but there is no others.
Here's an image: 
[image]

Happens with any theme.

let me know if I'm dull and just missed a setting or if I can do anything more.

<hr>

**Plugins installed**
- Z Probe Offset[z_probe_offset] v0.5
- **UI Customizer[uicustomizer] v0.1.5.8**
- Smart Preheat[smartpreheat] v0.0.6
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- PrettyGCode[prettygcode] v1.2.4
- OctoFlat Theme[octoflat] v0.2.0
- Network Health Monitor[networkhealth] v1.0.3
- LED Strip Control[LEDStripControl] v0.3.7
- GPIO Control[gpiocontrol] v1.0.8
- Firmware Updater[firmwareupdater] v1.10.10
- Filament Manager[filamentmanager] v1.7.2
- Detailed Progress[detailedprogress] v0.2.7
- Cost Estimation[costestimation] v3.2.0
- OctoEverywhere![octoeverywhere] v0.8.5
- OctoDash Companion[octodashcompanion] v0.0.3
- ipOnConnect[ipOnConnect] v0.2.4

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: false

**Software versions**
- OctoPrint 1.6.0
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:78.0) Gecko/20100101 Firefox/78.0

LazeMSS, 2021-05-10:
Maybe try disabling OctoFlat Theme plugin?

LazeMSS, 2021-05-15:
@Pen16  closing this due to no answer.

closed 2021-05-15 by LazeMSS