LazeMSS/OctoPrint-UICustomizer #155: Height of "body.UICResponsiveMode #settings_dialog_content" is not set

thelastWallE, 2021-05-13:
I would like to start a discussion on this css rule:
body.UICResponsiveMode #settings_dialog_content {
	height: calc(100% - 40px);
}
This is for the height of the settings tabs after substarcting the navpills on respmosive mode.
### **Before:**
[image]
overflow on the parent div->buttons etc don't stay on top.

### **after the css rule:**
[image]
overflow only on deeper div if set correctly from the plugin. Buttons staying on top.
As for now i need to set the css rule outside of my plugin to get the desired result.

LazeMSS, 2021-05-14:
Just to make sure - you wan't me to add:
```
body.UICResponsiveMode #settings_dialog_content {
	height: calc(100% - 40px);
}
```

?

thelastWallE, 2021-05-14:
This probably needs more testing. I had looked at a few plugins tabs and it was not breaking anything.
If the navpills have constantly the same height(40px) there wouldn't be a problem i think.
But yes i think it should be added, because it would leave the navpills on the top as you scroll down.
As it is now the overflown content causes the #settings_dialog_content to expand beyond the parent divs height and this would cause a scrollbar on the parent div.

thelastWallE, 2021-05-14:
The same with your own settings tab:
[image]
[image]
Would be better to be able to switch the tabs anytime.

thelastWallE, 2021-05-14:
Needs some css settings on the plugins css, too:
[image]

LazeMSS, 2021-05-15:
I get what you want to achieve but there are no rules in the current settings ie that says settings need to be built using tabs like this. I spent many days testing the current settings to be able to handle the many different plugins.

LazeMSS, 2021-06-28:
Closing - will not implement - its not part of the main way of the UI working.

closed 2021-06-28 by LazeMSS