LazeMSS/OctoPrint-UICustomizer #157: Spool Manager overlaps panel below

Xiondewah, 2021-05-27:
Updated to 0.1.5.9 which has fixed the layout of the edit/add spool manager - so thanks - but still have the problem where spool manager ui overlaps any panel below (so I've had to position it at the bottom of the left hand frame.

:-)

LazeMSS, 2021-05-27:
Screenshot?

Xiondewah, 2021-05-27:
Sorry - forgot!
[image]

LazeMSS, 2021-05-27:
Yes thats really not a bug in UI Customizer - that panel has a "nobr" tag - so the browser is told to never make linebreaks. This should be fixed in the spool manager plugin - no break is really no good for mobile/responsive layouts. As stated in https://github.com/OllisGit/OctoPrint-SpoolManager/issues/142 I will try and do a bit of revamp of the UI together with OllisGit.

closed 2021-05-27 by LazeMSS

Xiondewah, 2021-05-27:
This is frustrating because I posted it on OllisGit's hub and he said it should be with you - but thanks, hopefully the 2 of you will be able to sort it out between you.

OllisGit, 2021-05-27:
Hey @Xiondewah,

I closed your ticket, because it was duplicated. And I entered a link to the other ticket where LazeMSS is also involved in the discussion.
See your orig. Ticket:  https://github.com/OllisGit/OctoPrint-SpoolManager/issues/171

I never said that the issue must be solved by @LazeMSS !

OllisGit, 2021-05-27:
Btw. the br-element will be removed in the next release.

LazeMSS, 2021-05-28:
> Hey @Xiondewah,
> 
> I closed your ticket, because it was duplicated. And I entered a link to the other ticket where LazeMSS is also involved in the discussion.
> See your orig. Ticket: [OllisGit/OctoPrint-SpoolManager#171](https://github.com/OllisGit/OctoPrint-SpoolManager/issues/171)
> 
> I never said that the issue must be solved by @LazeMSS !

No panic @OllisGit  we will work out a good solution together ;)

Xiondewah, 2021-05-28:
Thank you both - the two plugins are so useful!