LazeMSS/OctoPrint-UICustomizer #158: filesFullHeight doesn'

EvilGremlin, 2021-05-27:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- Z Probe Offset[z_probe_offset] v0.5
- **UI Customizer[uicustomizer] v0.1.5.9**
- Top Temp[toptemp] v0.0.1.4
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- OctoPrint-WiFiStatus[wifistatus] v1.5.5
- SpoolManager Plugin[SpoolManager] v1.4.3
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- MultipleUpload Plugin[MultipleUpload] v1.1.0
- Firmware Updater[firmwareupdater] v1.11.0
- FileManager[filemanager] v0.1.6
- Exclude Region[excluderegion] v0.3.0
- Draggable Files[draggable_files] v1.0.2
- DisplayZ[displayz] v0.2.0
- DiscordRemote[discordremote] v2.69.0
- Custom Control Editor[customControl] v0.2.2
- Camera Settings[camerasettings] v0.1.2
- Active Filters Extended[active_filters_extended] v0.1.0
- M73 Progress Plugin[m73progress] v0.2.1
- Consolidated Tabs[consolidatedtabs] v1.0.1
- Bed Visualizer[bedlevelvisualizer] v1.0.1
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Action Commands[actioncommands] v0.4

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: false
- compressTempControls: false
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true
- themeLocal: true

**Software versions**
- OctoPrint 1.6.1
- Python 3.8.7

**Browser**
- Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.212 Safari/537.36 OPR/76.0.4017.154

LazeMSS, 2021-05-29:
Can you provide a bit more info. In the latest update its turned into a resizable for files and term

EvilGremlin, 2021-05-30:
Ohhh, i searched for resize corner in wrong place...

closed 2021-05-30 by EvilGremlin

hogfanboy, 2022-07-19:
found the resize  corner  after  15 min of frustration.  ( it is at the bottom of  the vertical  scroll bar) . I was  thinking it was in the bottom right corner of the file widget/box/window.  might consider adding a little more verbiage where to find it

LazeMSS, 2022-07-19:
> found the resize corner after 15 min of frustration. ( it is at the bottom of the vertical scroll bar) . I was thinking it was in the bottom right corner of the file widget/box/window. might consider adding a little more verbiage where to find it

It's the browser way of doing it. I can't change it

hogfanboy, 2022-07-19:
Thanks for the quick comment back.  and thanks for the plugin.
I was  just  thinking, for those of us that look at the readme on the
github site,  just add an extra line on the description of that option
that says "the resize button is at the bottom of the vertical scroll bar".
Thanks again  I was attempting to give some feedback with a solution.

KBK