LazeMSS/OctoPrint-UICustomizer #160: No webcam stream after some minutes of print

Gil80, 2021-06-19:
## Description
Before using UI Customizer, I could run the webcam that is attached to my RPi running Octopring, for hours and hours.
After installing UI Customizer, the stream will fail after some minutes in the worst case, or an hour in the best case.

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.5.9**
- Access Anywhere - The Spaghetti Detective[thespaghettidetective] v1.6.7
- Detailed Progress[detailedprogress] v0.2.7
- Auto Terminal Input[AutoTerminalInput] v0.1.0
- OctoEverywhere![octoeverywhere] v1.0.0
- M73 Progress Plugin[m73progress] v0.2.1
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.26.0
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: true

**Software versions**
- OctoPrint 1.6.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:89.0) Gecko/20100101 Firefox/89.0

LazeMSS, 2021-06-22:
Which stream fails - the main or the extra widget?
What type of stream are you using?
I haven't had any other reports of this and my own is running 24-7 so please provide as much info as possible

Gil80, 2021-06-22:
If the mainstream fails, so does the extra widget. Either way, they both fail.

Can you please explain what you mean by "type of stream"?

I use the C920 Logitec webcam and I also tried a two types of 720p Microsoft webcam: 1- https://www.bhphotovideo.com/c/product/744410-REG/Microsoft_6CH_00001_LifeCam_Cinema_WebCam_For.html

2- https://www.flipkart.com/microsoft-lifecam-hd-3000-webcam/p/itmdvav7demybtgd

This issue happened with all three cams.

I decided to edit the octopi.txt file and uncomment #camera="auto" and change it to camera=USB
In addition, I edited the following when using the C920 Logitech webcam:
`camera_usb_options="-r 1920x1080 -f 30"`
so far I was able to have an 11 hours print without a webcam stream failure.

I'm not sure if these changes helped at all and if yes, then why? why doesn't the default setting work.

LazeMSS, 2021-06-22:
I wanted to know if you are using mjpg or HLS - but i'm assuming you are using the default which is mjpg.

It might be that the fallback (auto) option can't handle having two streams at the same time but I doubt it - as I said I have never had any report of this and im also using the same camera myself without any issues.

Btw. if you have the problem again try running: `vcgencmd get_throttled` it might be a power / cpu issue when running dual stream output but again I doubt it but nice to check

Gil80, 2021-06-22:
By dual stream, you refer to two different webcams at the same time? or two streams of the same webcam at the same time?

LazeMSS, 2021-06-22:
Two streams of the same webcam at the sametime - the RPI needs to serve them etc.

What RPI do you have?

Gil80, 2021-06-22:
RPi3 B.
Powered with the official RPi adapter.
It doesn't have undervoltage/overheating issues.

LazeMSS, 2021-06-26:
> RPi3 B.
> Powered with the official RPi adapter.
> It doesn't have undervoltage/overheating issues.

Very strange - is it still a problem?

Gil80, 2021-06-26:
Once I set to explicitly use USB instead of AUTO, I no longer have issues with the Logitec C920.
I'll switch webcams and see if this happens with others as well.

LazeMSS, 2021-06-26:
Thanks - I will also test on my installation. Basically UI Customizer uses the exact same stuff as the normal webcam so there should be no difference with or without it installed in stability of the webcam feed

LazeMSS, 2021-06-30:
I can't reproduce 🙈

LazeMSS, 2021-07-10:
Will close and reopen if other reports similar

closed 2021-07-10 by LazeMSS