LazeMSS/OctoPrint-UICustomizer #161: Touch UI Compatibility

RagingHog55, 2021-06-29:
So this could be just me, but when viewing Octoprint on a desktop/laptop the TouchUI menu is in the middle of the top bar along with Progress: 0% Layer: - of - Height: - of -mm. However when i go over to my raspberry pi 7" touchscreen next to the printer this is no longer there. Is there a way to get this functionality on a smaller screen?

LazeMSS, 2021-06-30:
The problem is touch UI and UI customizer collides.

When running it on Touch screen always add: #touch to the URL. So the URL to use on the 7" Touch is : http://IP of RPI#touch for example http://192.168.1.1/#touch

This hopefully fix it.

But I would encourage you to use one of the dedicated UIs for the Touch screens like OctoDash and remove touch UI.

RagingHog55, 2021-06-30:
Thank you, i hadn't looked for alternatives to TouchUI in quite sometime. I like your UI and just switched to it and noticed touchUI no longer worked. I am in the process of getting OctoDash installed! Thanks!

LazeMSS, 2021-06-30:
Great closing now then b

closed 2021-06-30 by LazeMSS