LazeMSS/OctoPrint-UICustomizer #170: UI-Customizer running on WebView?

Jyers, 2021-07-16:
I was wondering if there was any way to get UI Customizer to play nice with chromium webview? I have a very strange android tablet that is pretty much purpose built just to run webview... It can run a full browser, but it's not the best experience. 

Is there any setting I can configure on the browser side that might help compatibility? Or is this just something that will most likely never function due to the nature of this plugin?

LazeMSS, 2021-08-14:
It should do that but i can't test - what's the problem?

Jyers, 2021-08-15:
@LazeMSS It just doesn't use the theme whatsoever. It goes back to default octoprint. If there is any way I can help/test to get this working I would be happy to :)

LazeMSS, 2021-08-17:
Maybe its related to the webview - is there any easy way to test t?

Jyers, 2021-08-19:
The first few of [these](https://www.makeuseof.com/tag/lightweight-android-browsers/) all use webview, so you can take your pick

LazeMSS, 2021-08-19:
Tried the via browser on my Samsung mobile.

Seems to working as expected. Maybe the webviewer on your tablet is not updated.
[image]

[image]

LazeMSS, 2021-08-20:
@Jyers ?

Jyers, 2021-08-20:
Interesting, I'll try it when I get home today

LazeMSS, 2021-08-20:
@Jyers great let me know - I know there are some devices that don't get the latest webview - but maybe it can be sideloaded: https://www.apkmirror.com/apk/google-inc/android-system-webview/

LazeMSS, 2021-08-22:
@Jyers  any news?

Jyers, 2021-08-22:
I also was able to get the proper UI on my android device, but even after updating webview on the tablet I have not had any luck getting the UI to show up properly. It definitely seems like it is an individual issue of some kind with my device. Unfortunately I'm getting ready for a move right now so I don't have time to do a bunch of troubleshooting. But when I have time to try and figure it out I will update this in case it could help others.

LazeMSS, 2021-08-22:
Okay will close - you can reopen if you find anything new.
Good luck moving :)

closed 2021-08-22 by LazeMSS