LazeMSS/OctoPrint-UICustomizer #171: Theme doesn't stay set on dual instance

HWCronicus, 2021-07-18:
Since your most recent update I've been having a problem with having 2 different theme choices on 2 different octoprint instances running on a single Pi. It used to work just fine. So for example one instance would have the orange theme, and the other instance would have a blue theme. And they used to stay set after reboots, windows closing, and stuff like that. But now If I close the browser tab for either one there's a 50/50 chance that it would change its theme to the other instance's theme. It doesn't prefer one instances theme over the other. It's just random on which one of the 2 set themes will take over both instances. Now on the instance that has the wrong theme, the theme selection menu will show that the right theme is selected. And to get it back to the original set theme, I have to switch the theme completely and them switch back to the original theme. Not sure what I can do to fix it. It's nothing game breaking just inconvenient to have to keep switching back. I know I could just use one theme and be good. But if it used to work idk why it still can't work. But this is a great plugin and even if it stays like this I'll definitely still use this plugin lol. Keep up the great work, ty.

Jyers, 2021-07-20:
How do you have you're instances set up? I have three on one pi and have not seen any issues like this. For my instances they each have their own virtual environment and their own base directory, so they can't share any files unless I specifically add symlinks between them. Perhaps your instances are set up in a way where they are sharing some data files.

HWCronicus, 2021-07-20:
I'm not a 100% sure on this but I do believe that are sharing main plugin files between each instance but not stored data. I followed a guide by Chris Riley on his YT channel (https://youtu.be/FIPbiImixZM) to set up the dual instance OctoPi. But other plugins and saved data specific to each instance are not shared between the 2, and this is a new problem that wasn't present before the update. If you have a link or guide on how to setup each in a virtual environment I'll give that a shot. I don't really want to have to pretty much start from scratch again but no worries if that's what gets it back to normal. Ty.

Jyers, 2021-07-20:
Here's the guide I used: https://makenotes.de/2020/03/octoprint-on-ubuntu-using-python3-and-systemd/

If you backup your instances from within octoprint it makes it super simple to restore them after the switch btw ;)

HWCronicus, 2021-07-24:
I'll give it a shot. Thanks

LazeMSS, 2021-08-14:
I think @Jyers  solution is right :)

closed 2021-08-14 by LazeMSS