LazeMSS/OctoPrint-UICustomizer #172: [Feature Request] True full screen camera

ManuelMcLure, 2021-07-25:
It would be really nice if the zoomed/floating camera had an icon I could click to make it truly full screen. Currently to achieve this I have to install the FullScreen Webcam plugin and uncheck "Hide webcamera on Control tab" so I can use that plugin to achieve true full screen webcam.

LazeMSS, 2021-07-25:
On the to-do

LazeMSS, 2021-08-17:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.6.2
Please check and reopen if not fixed/working as expected.

electronicm, 2021-08-22:
With a 4:3 webcam, the lower part is cut off in landscape.

[image]
[image]

LazeMSS, 2021-08-22:
What resoultion is the webcam?

electronicm, 2021-08-22:
960 x 720

LazeMSS, 2021-08-22:
Thanks I think I have it fixed - you can try and add this to your custom css under advanced:
```
#UICWebCamFull img,#UICWebCamFull video{
    max-height: 100vh;
    max-width: 100vw;
}
```

And let me know it that works :)

electronicm, 2021-08-23:
> #UICWebCamFull img,#UICWebCamFull video{
>     max-height: 100vh;
>     max-width: 100vw;
> }

This solves the problem

LazeMSS, 2021-08-31:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.6.6
Please check and reopen if not fixed/working as expected.

closed 2021-08-31 by LazeMSS