LazeMSS/OctoPrint-UICustomizer #174: No fixed header on mobile/responsive

electronicm, 2021-08-10:
## Description
If I activate the two options

- Improve mobile/responsive
- Fixed header/topbar"

the header does not stay at the top and disappears when scrolling.

<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.6.1**
- Top Temp[toptemp] v0.0.1.4
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Simple Filament Change Buttons[simple_filament_change_buttons] v1.1
- Tempsgraph Plugin[tempsgraph] v0.3.7
- Printer Stats[stats] v3.1.2
- Slicer Print Time Estimator[SlicerEstimator] v1.0.3
- RequestSpinner[requestspinner] v0.2.0
- Print History Plugin[printhistory] v1.3
- MeatPack[meatpack] v1.5.21
- GcodeEditor[GcodeEditor] v0.2.12
- Exclude Region[excluderegion] v0.3.0
- Octoprint Display ETA[display_eta] v2.0.2
- Camera Settings[camerasettings] v0.2.2
- BLTouch Plugin[BLTouch] v0.3.4
- BetterHeaterTimeout[BetterHeaterTimeout] v1.3.0
- Marlin EEPROM Editor[eeprom_marlin] v3.1.0
- M73 Progress Plugin[m73progress] v0.2.1
- Bed Visualizer[bedlevelvisualizer] v1.1.0
- Backup Scheduler[backupscheduler] v0.1.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: false
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: false

**Software versions**
- OctoPrint 1.6.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Linux; Android 11; MI 9) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Mobile Safari/537.36

LazeMSS, 2021-08-14:
Works on my android devices and apple devices - can you provide a screenshot?

electronicm, 2021-08-15:
LANDSCAPE: Header is fixed
https://user-images.githubusercontent.com/13643227/129479357-91949452-e1ff-4430-8732-c1b7882a2bcd.mp4

PORTRAIT: Header is not fixed
https://user-images.githubusercontent.com/13643227/129479365-ba79e6fd-6152-4888-8c2a-763bfab751f1.mp4

LazeMSS, 2021-08-15:
Thanks. Yes that's right on mobile the fixed header and footer is not fixed. It's part of the standard bootstrap behaviour. I will try and fix it

electronicm, 2021-08-16:
I have created PR #178 for this problem.

LazeMSS, 2021-08-17:
Not implemted yet so keeping it open

LazeMSS, 2021-08-17:
https://github.com/LazeMSS/OctoPrint-UICustomizer/commit/8f7aa32963fae5bdbb86ce2bae0c3174f90a13d2

LazeMSS, 2021-08-17:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.6.2
Please check and reopen if not fixed/working as expected.

electronicm, 2021-08-18:
Thanks for the quick support and sorry for my confusing approach. Is the first time I'm contributing in github and still need to get familiar with it. :D

The header is now fixed.
Unfortunately the header now overlaps the content area and in the portrait  
all icons are now in the second line below the OctoPrint logo.

The sliding down of the icons causes the following code in bootstrap-responsive.css:
```
@media(max-width:979px) {
- .nav-collapse {
- clear:both
- }
}
```

To prevent content overlay, I added the following to bootstrap-responsive.css:
```
.octoprint-container {
   margin-top: 50px;
}
```
A little more work is needed here though, as this entry is so for fixed and not-fixed headers.

Screenshots/-records will follow as soon as my battery is recharged.

LazeMSS, 2021-08-19:
Yes it hard to fix - I have WIP solution that scrolls the icons when there is not enough room in the top menu and an a alternative version where they are placed in a dropdown.

LazeMSS, 2021-08-19:
I have a couple of fixes on the way now

LazeMSS, 2021-08-20:
Should be improved in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.6.3
Please check and reopen if not fixed/working as expected.****

closed 2021-08-20 by LazeMSS

electronicm, 2021-08-22:
Partially it works.
I have created a small test matrix of conditions that I could identify:

[image]

I found the problem with the MainMenu Background in the "Discorded" theme. It has the color code #ebebeb.
I have not tested other themes in this respect.

LazeMSS, 2021-08-22:
I will try and look into it.