LazeMSS/OctoPrint-UICustomizer #176: Broken Info popup layout (OctoPrint 1.7.0rc2Python 3.7.3OctoPi 0.18.0)

LMS0815, 2021-08-12:
The layout of the Information (bell) screen seems to be broken.

[image]

Tool popup seems to normal.

[image]

[octoprint-systeminfo-20210812124320.zip](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/6975184/octoprint-systeminfo-20210812124320.zip)

LazeMSS, 2021-08-14:
Will check asap

LazeMSS, 2021-08-14:
Ahh this sometimes happens due the new modal dialogs - I will try and see if I can make a fix together with the main octoprint

LazeMSS, 2021-08-14:
Ohh have you tried clearing your browser cache ?

LMS0815, 2021-08-14:
Clearing the cache seems to fix the issue. (Although I see the small window which is resized on a slow system).

Thank you for the fast reply/solution,

LMS0815, 2021-08-17:
Refresh only solves the issue on some systems (with 4:3 Monitor).
The issue seems to be related to the 16:9 Screen. 
If I move the window to my second 4:3 Monitor, the window is resized to the desired Size.
Moving it back keeps the size. After reloading same issue.

Same wit Edge and Chrome...

LazeMSS, 2021-08-17:
This is not related to the UI Customizer plugin - it still happens if I disable it.

closed 2021-08-17 by LazeMSS