LazeMSS/OctoPrint-UICustomizer #177: Gcode Viewer aspect ratio doesn't maintain proper proportions

crzy1, 2021-08-13:
## Gcode Viewer aspect ratio doesn't maintain proper proportions
**Gcode Viewer will not show proportionally, warps image.
Depending on how the web browser is sized, the gcode image gets squished vertically or horizonatally
This is not intermittent, it will happen all the time
Tried reinstalling plugin**
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.6.1**
- Top Temp[toptemp] v0.0.1.4
- Wemo Switch[wemoswitch] v0.1.10
- Procastinator[procastinator] v1.1.1
- OctoText[OctoText] v0.3.1
- Malyan/Monoprice Connection Fix[malyan_connection_fix] v0.1.3

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- themeLocal: false

**Software versions**
- OctoPrint 1.6.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36

LazeMSS, 2021-08-14:
Can you provide a screenshot?

crzy1, 2021-08-15:
[image]

crzy1, 2021-08-15:
Notice the grid is not proportional.  It's elongated vertically.  if I widen the window, it will be elongated horizontally.  A 1:1 aspect ratio should be locked in as the window is sized.

LazeMSS, 2021-08-15:
Yes I will try and fix it

crzy1, 2021-08-16:
Thank you, it would be greatly appreciated.

Louie V. Lefty
Senior Network Systems Engineer CNE CCNA
Business email: ***@***.***
Personal email: ***@***.***
847-461-9755

"Live free or die, death is not the worst of evils."
General John Stark

LazeMSS, 2021-08-17:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.6.2
Please check and reopen if not fixed/working as expected.

crzy1, 2021-08-18:
Sorry, but I'm still seeing the issues on all of my computers.  It happens in the chrome browser and the fire fox browser.  The UI Customizer doesn't even show up when using IE...  If you drag the browser edge in and out you'll dynamically see the grid width get narrow and wide as you drag.  The same thing happens in the smaller view of the code view too,  Sorry to drag this on

LazeMSS, 2021-08-18:
The widget is just a scaled (zoom) version of the main gcode viewer so it should behave the same as that one. Could you tell me what resolution your running the browser in and maybe also the gcode as sample

crzy1, 2021-08-18:
This happens on any computer/display.  One is at 1920x1080,  one is at
1280,1024 and two others are at 1024x768.  Tray dragging/resize a chrome
browser from the right or left edge and you will see the grid narrow and
widen.  It seems that the aspect ratio is not locked in to stay
proportional.  Like in the old days of using iFrames in HTML.

I hope this helps.

Louie V. Lefty
Senior Network Systems Engineer CNE CCNA
Business email: ***@***.***
Personal email: ***@***.***
847-461-9755

"Live free or die, death is not the worst of evils."
General John Stark

LazeMSS, 2021-08-19:
It should be using the same width and height ie always a square after the latest update

crzy1, 2021-08-19:
I'm really sorry for beating this issue, but are you seeing the same thing
that I'm seeing when resizing the window?

Louie V. Lefty
Senior Network Systems Engineer CNE CCNA
Business email: ***@***.***
Personal email: ***@***.***
847-461-9755

"Live free or die, death is not the worst of evils."
General John Stark

crzy1, 2021-08-19:
Another thought would be to not allow the column with the gcode to resize
dynamically, keeping it's aspect ratio 1:1.  Once you set the size of that
column it would be stationary.

Louie V. Lefty
Senior Network Systems Engineer CNE CCNA
Business email: ***@***.***
Personal email: ***@***.***
847-461-9755

"Live free or die, death is not the worst of evils."
General John Stark

LazeMSS, 2021-08-19:
No I really can't seem to recreate or maybe understand what you are seeing. 

My plugin allows the entire UI to be responsive so it makes little sense to have a fixed item.
The gcode widget should be 1:1 aspect ratio.

crzy1, 2021-08-19:
Well that makes it tough.  It happens on all the PCs I use and on the 7"
displays on the PIs.  Below are two more pics, maybe I'm not saying things
correctly.  In the first pic I narrowed my browser and the grid is narrow
and tall. In the second pic, I widened my browser and the grid is properly
displayed with a 1:1 ratio.  When maximizing the display to the extent of
my PC monitor things are good. But once I restore the window to a smaller
size the grid is once again distorted.  The third pic show that no matter
how much I size the window vertically the height of the grid stays the
same, unlike when sizing the window horizontally.  So sizing the window
vertically doesn't change the grind size vertically, but sizing
horizontally sizes the grid size horizontally causing the aspect ratio to
be off.  Vertically it seems to be locked in, but horizontally it is
dynamic.

I just went back and tested every option for the customizer.  This only
happens when the "Gcode full-size" option is not selected.  I don't know if
it's a feature.  When I select 'Gcode full-size" the aspect ratio stays at
1:1.  I don't know if this is something that needs fixing or if it lead you
to checking something out.  But I can just keep that option selected.

I really appreciate your time.
Let me know what you think.

[image: image.png]

[image: image.png]
[image: image.png]

[image: image.png]

Louie V. Lefty
Senior Network Systems Engineer CNE CCNA
Business email: ***@***.***
Personal email: ***@***.***
847-461-9755

"Live free or die, death is not the worst of evils."
General John Stark

LazeMSS, 2021-08-19:
Pictures are missing :/

LazeMSS, 2021-08-20:
Ohh and yes the base aspect is copied from the main gcode viewer. If that is not in 1:1 then the small widget is also not in 1:1

LazeMSS, 2021-08-20:
Should be improved in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.6.3
Please check and reopen if not fixed/working as expected.

closed 2021-08-20 by LazeMSS

crzy1, 2021-08-20:
Thanks for working so hard on this.  However it didn't fix the problem.
However when I do select "Gcode full-size"  in the General options for UI
Configurator.  That keeps the aspect ratio properly applied.

So, with "Gcode full-size" selected, it keeps the entire gcode grid borders
within the browser window and keeps the aspect ratio consistent.   When
"Gcode full-size" is unchecked, it allows the gcode view to shrink
horizontally but allows it to stay full size vertically going past the
bottom of the browser border.

After studying the option, I think that is why the "Gcode full-size" option
is there.  There isn't any way to keep the aspect ratio the same and allow
the gcode view to flow past the bottom of the browser window.

I think it's all working properly, and in order to keep the aspect ratio
the same the "Gcode full-size"  option must be selected.  Bottom line.

[image: image.png]

Louie V. Lefty
Senior Network Systems Engineer CNE CCNA
Business email: ***@***.***
Personal email: ***@***.***
847-461-9755

"Live free or die, death is not the worst of evils."
General John Stark

Thank you for all of your work.
I think things are working fine.

LazeMSS, 2021-08-20:
Great thanks for the feedback