LazeMSS/OctoPrint-UICustomizer #183: Tabs display 3rd party plugins first after closing settings

kantlivelong, 2021-08-20:
For some weird reason tabs display 3rd party plugins first after hitting close or save in settings even if no changes were made.. Refreshing the UI resolves. I am not using any custom ordering.

Not seeing any useful info in JS console.

Expected:
- Temperature
- Control
- GCode Viewer
- Terminal
- Timelapse
- [MORE DROPDOWN]
- Bed Visualizer
- EEPROM Editor

Current:
- Bed Visualizer
- EEPROM Editor
- Temperature
- Control
- GCode Viewer
- [MORE DROPDOWN]
- Terminal
- Timelapse

Config:
```
  uicustomizer:
    addWebCamZoom: false
    centerTopIcons: false
    compactMenu: false
    compressTempControls: false
    customCSS: "/*\n#term .terminal #terminal-output,#terminal-output-lofi {\n   \
      \ min-height: 600px;\n}\n\n#term .terminal #terminal-output, #term .terminal\
      \ #terminal-output-lowfi {\n    height: 500px;\n    resize: vertical;\n    max-height:\
      \ inherit;\n}\n*/"
    filesFullHeight: false
    fixedFooter: false
    fixedHeader: false
    fluidLayout: false
    hideGraphBackground: false
    mainTabs:
    - - temp_link
      - true
      - false
      - fas fa-thermometer-half
      - iconOnly
      - false
    - - control_link
      - true
      - false
      - fas fa-expand-arrows-alt
      - iconOnly
      - false
    - - gcode_link
      - true
      - false
      - fab fa-codepen
      - iconOnly
      - false
    - - term_link
      - true
      - false
      - fas fa-terminal
      - iconOnly
      - false
    - - tab_plugin_bedlevelvisualizer_link
      - true
      - false
      - fas fa-th
      - iconOnly
      - false
    - - tab_plugin_eeprom_marlin_link
      - true
      - false
      - fas fa-microchip
      - iconOnly
      - false
    - - timelapse_link
      - true
      - false
      - fas fa-film
      - iconOnly
      - false
    mainTabsCustomize: false
    mainTabsIconSize: fa-2x
    navbarplugintempfix: false
    responsiveMode: false
    rows:
    - _00#connection_wrapper: true
      _01#state_wrapper: true
      _02#files_wrapper: true
      _03#sidebar_plugin_firmware_check_info_wrapper: false
      _04#sidebar_plugin_firmware_check_warning_wrapper: false
      _05#sidebar_plugin_action_command_notification_wrapper: true
      _06#UICWebCamWidget: false
      _07#UICGcodeVWidget: false
      _08#UICTempWidget: false
    - _00div.UICmainTabs: true
    - {}
    theme: discorded
    themeLocal: false
    themeVersion: 0.2.4
    topIconSort:
    - navbar_plugin_action_command_prompt
    - navbar_plugin_DisplayLayerProgress
    - navbar_plugin_navbartemp
    - navbar_plugin_lightcontrol
    - navbar_plugin_psucontrol
    - navbar_plugin_ups
    - navbar_plugin_logging_seriallog
    - navbar_plugin_logging_plugintimingslog
    - navbar_plugin_pi_support
    widths:
    - '4'
    - '8'
    - '0'
```

LazeMSS, 2021-08-20:
You dont have the "Customize tabs" checked?

kantlivelong, 2021-08-20:
Went to recheck. "Customize tabs" is unchecked.

Here's a capture: https://imgur.com/a/MJiMBxq

LazeMSS, 2021-08-20:
ahh now i see

LazeMSS, 2021-08-20:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.6.3
Please check and reopen if not fixed/working as expected.

closed 2021-08-20 by LazeMSS

kantlivelong, 2021-08-20:
Success! 👍